prusa3d/Prusa-Firmware-Buddy #3630: [BUG] Canceling object taller than other objects causes weird result

mtaumike, 2023-12-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - [5.1.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint** - [USB]

**Describe the bug**
  Cancelling the taller object causes the printer, once the other 2 objects are complete, to slowly lift Z while also retracting filament. I tested this twice with different slices once this occured the first time (image attached) and was able to reproduce it.

  It is also kind of loud while this is going on as the extruder makes noises like it is pushing and retracting but it is just retracting.

**How to reproduce**
  Have multiple objects on plate where one object is taller than the others then cancel that object. Once the shorter objects are complete this behaviour occurs.

**Expected behavior**
  Print finishes without unloading filament

**G-code**
  I'd suggest slicing your own

**Image**
[image]

mtaumike, 2024-01-01:
Has anone else been able to recreate this? I can do it - even did a factory reset (after grabbing my sheet offsets that is) and sliced up new tests (just some cubes, 2 shorter, 1 taller) and can confirm, on my gear, that this occurs

miohna, 2024-03-01:
I can confirm this issue.
In my case (Mini+, 5.1.2+13478) it is not even linked to the height of the object (tested with two objects of same height).
Once I cancel the object being printed, the Mini+ performs the same move as described by @mtaumike : lifting Z-Axis while retracting the filament at the same time. It feels like the printer is skipping all G-code commands but the filament retraction. Same issue appears if I cancel the second object right after.

This issue renders the cancel object function unusable for the Mini+ as the remaining objects are printed without filament extrudes at the beginning (due to the filament being retracted beforehand).

mtaumike, 2024-03-02:
Finally someone else is seeing this! WHOOP!

But I should not be excited but would love to see this fixed. Thankfully I don't have a need to cancel objects that often but right now if I know I need to I just cancel the whole print job.

miohna, 2024-03-04:
I have to specify that the issue is only linked to sequential printing.
If using the cancel object feature for a non-sequential print, it does work as expected in my case.

mtaumike, 2024-03-04:
Interesting as it happens to me regardless of sequential or just normal plate printing.

danopernis, 2024-06-07:
@miohna @mtaumike sorry for taking so long to respond. Can you confirm if the bug is present in lastest firmware?

miohna, 2024-06-07:
Further evaluation on my end has show that the issue was linked to a bug in the Prusa Slicer - resolved now.