prusa3d/Prusa-Firmware-Buddy #3631: MK4: 5.1.2 Cancel Object not selectable on menu[BUG]

bryn51, 2023-12-21:
Searched, did not see it.

**Printer type** - [Mk4

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - Enclosure

**USB drive or USB/Octoprint**
  PrusaLink

** Slicer Version **
Prusa Slicer 2.7.1  on Macbook Pro M1 running Sonoma MacOS 14.1.2

**Describe the bug**
  The "Cancel Object" command on "Control" menu is grey'ed out and not available for selection when :
  a) 3 identical objects are defined on the bed as Instances.
  b) when the Instances are altered by Prusa Slicer to set as separated instances: 3 objects all of the same name.
Prusa Support Reference: Rodolfo

**How to reproduce**
  Just look at the attached PS Project file, slice it, and load onto the printer.

**Expected behavior**
 The menu option should not be greyed out, it should be selectable, and the functionality of Cancel Object should work.

**G-code**
I got the 3MF file OK ?

**Crash dump file**
It did not crash.

**Video**
Video is unnecessary.
[Hawthorn  HoboPendant Sleeve (x3) PLA.3mf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13735290/Hawthorn.HoboPendant.Sleeve.x3.PLA.3mf.zip)

Harry-58, 2023-12-21:
I had the same problem.
In my case it was due to the slicer settings.
Print settings - Output options - Label objects must be set to "Firmware specific".

bryn51, 2023-12-21:
It may be correct, I'll check it and see. 
But even so, seems like the preconditions needed to have this work could have/should have been stated in release notes, as there is no other documentation.

PLhoas, 2023-12-22:
Thanks Harry-58, it solved it for me.  I owe you a Belgian beer.

NicholasRakita, 2024-02-09:
Just to add some clarity here, it's PrusaSlicer that inserts some special GCodes that help to delineate the objects during printing which makes "cancelling" object mid-print possible.

More likely than not, this is a PrusaSlicer issue and not a Firmware problem (at least I hope not :crossed_fingers:)

WTPascoe, 2024-02-13:
I have the same issue, and Label Objects is set to Firmware specific in slicer. I'm on 2.7.1

closed 2024-07-01 by github-actions[bot]