prusa3d/Prusa-Firmware-Buddy #3633: [BUG] no NGEN filament type

metebalci, 2023-12-21:
**Printer type** - [MK3.9]

**Printer firmware version** - [5.1.2]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
USB flash drive

**Describe the bug**
There is no NGEN filament option in the printer but there is in the slicer. When a gcode with NGEN filament type is sent, the printer always issues a warning since NGEN filament has to be loaded as PETG etc.

**How to reproduce**
Choose an NGEN filament in the Slicer, load NGEN filament (but select PETG as there is no NGEN option). The printer issues filament type mismatch warning.

**Expected behavior**
It should be possible to choose an NGEN filament type in the printer.

bkerler, 2023-12-21:
Hi,
in order to write a PR for your request, can you please
a) confirm that NGEN works on the MK3.9 without issues / modifications
b) the temperature profile you used (recommended values for NGEN are nozzle=240, heatbed=90) ... you can try the PP (prefered but with heatbed set to 90 instead of 100) for example

Thanks

metebalci, 2023-12-21:
I just started trying an NGEN filament and my understanding is it is like PETG so I do not think there is a major problem, it prints as expected. I used ColorFabb nGen profile in the Slicer.

bkerler, 2023-12-21:
Ok, it's also officially listed on the Prusa homepage, so the PR should be fine then. Still let me know if you hit any problems with temp (sticking to the bed or stringing).

bkerler, 2023-12-21:
Oh and please change the subject from bug to feature request. Thanks :)

bkerler, 2024-01-02:
I added a new PR which allows usage of custom filament profiles, which is even more convenient :)

metebalci, 2024-01-03:
> I added a new PR which allows usage of custom filament profiles, which is even more convenient :)

Great!

closed 2024-07-01 by github-actions[bot]