prusa3d/Prusa-Firmware-Buddy #3637: Prusa XL - Filament Sensor Runout Option

Inspiron41, 2023-12-25:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [Prusa XL ]
**Optional upgrades** - [e.g. Filament Runout Sensor]

**Is your feature request related to an existing problem? Please describe.**
Excessive filament wasted with using just the side filament sensor

**Describe the expected functionality**
 I would like to see an option in the menu to let users pick which filament sensor to use. I prefer to nextruder filament sensor.  For me it's easier to remove the PTFE tubing and remove it there.

mtaumike, 2024-01-01:
I think this is a good idea. The 1.5 meter of PTFE tubing leaves a lot of filament to toss

hickengr, 2024-01-08:
I would also like this feature for the same reason as mtaumike.

hickengr, 2024-06-06:
I am using XL firmware 6.0.1 and the side runout sensor can be turned off.  However, when I tried it, the print did not stop immediately when the tool filament sensor detected the runout and it kept trying to print after the filament was no longer being extruded.  This caused part of the infill to not be printed.  It was ok for that specific print.  However, if it happens on a perimeter, it could cause an unwanted blemish.  Can this feature be tweaked further to immediately pause the print and switch to the joined tool?

closed 2024-08-13 by github-actions[bot]