prusa3d/Prusa-Firmware-Buddy #3639: [BUG] 0.6 nozzle sporadic underextrusion

bdynamic, 2023-12-26:
**Printer type** -  MK4

**Printer firmware version** - 5.1,2

**Original or Custom firmware** -[Original/

**Optional upgrades** - 0.6 Nozzle

** Used USB drive

**Describe the bug**
  When using the 0.6 Nozzle there is a sporadic underextrusion in the first layer (see photos). This bug existed at least in the last two FWs. If the error occures it seems to persist. Reset of printer and restarting the very same GCODE normally solved the problem. If the problem exists there is no difference in printing via Octopring or from USB. The problem strength seem to differ. Sometimes there is even a visible gap between the outer printlines.

**How to reproduce**
  unknown. Sporradic occurence

**Expected behavior**
  Always a good print

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
The Crashdump file was generated after the reset (when everything was fine again). A new one can be provided as soon as the problem occures again.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

[Remote_holder.3mf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13769735/Remote_holder.3mf.zip)

Bad example:
[image]

Good print, same GCODE:
[image]

TheMoen, 2024-01-21:
I had this issue yesterday, It appears to be from oozing after the last print etc..
The reason it is an issue is that both cleaning and probing with the nozzle happens at 175°C. PETG simply is not soft enough at that teperature, so the printer belive the printbed is higher than it actually is. Leading to incorrect "first layer calibration". make sure the nozzle is clean if you are having this issue.

Not sure if this is a slicer setting or firmware..

alecharp, 2024-02-25:
Thank you @TheMoen for the explanation. I just got that problem and you are right, I'm only seeing this with PETG.

> The reason it is an issue is that both cleaning and probing with the nozzle happens at 175°C. PETG simply is not soft enough at that teperature, so the printer belive the printbed is higher than it [...]

But in that case, couldn't the "nozzle cleaning" step be done at a different temperature, higher for filament requiring it? That could be done in the FW, no? 

I would like to add that I clean the nozzle, pre-heating (to not mess with the auto leveling / pushing on the nozzle). However, it is possible that some PETG filament "leak" through the nozzle when it comes to temperature. So it gets soft enough to leak, but not enough to be cleaned correctly before the leveling.

Eriobis, 2024-05-03:
I am seeing the same kind of issue, but with PLA.
Firmware 6.0.0 using 0.6 nozzle.
PETG + textured bed  always work fine but with PLA and PEI smooth sheet I get nasty result for the 1st layer. :(

closed 2024-07-15 by github-actions[bot]