prusa3d/Prusa-Firmware-Buddy #3640: [FEATURE REQUEST] Set Ready via REST API

Skaronator, 2023-12-27:
**Is your feature request related to an existing problem? Please describe.**
As of 5.1.0 it is possible to set the printer ready from the printer UI itself. It would be great if that feature would be available via the REST API as well.

**Describe the expected functionality**
API endpoint that:
* Set Printer Ready
* Unset Printer Ready
* Include Ready Status / not Ready Status in /api/v1/status

https://github.com/prusa3d/Prusa-Link-Web/blob/master/spec/openapi.yaml

Skaronator, 2024-06-10:
Not stale...

Tomcus, 2024-06-10:
This is valid feature request. I think we could add it @JakoobCZ.

Zero-Circle, 2024-06-27:
This issue is the same as this one i made this week. #4058

nvtkaszpir, 2025-01-14:
Currently firmware allows setting printer status to ready via menu or processing commands coming from the PrusaConnect.

It would be very useful if this was also available via REST API in PrusaLink and gcode (as MK3 processes gcode `M72`).

arvacode, 2025-07-28:
I would really benefit from at exposing the `ready` status via **/api/v1/status**.
I'm surprised this hasn't been implemented into prusalink web to mirror prusaconnect.