prusa3d/Prusa-Firmware-Buddy #3641: Disabled heating on Prusa Mini leads to nozzle blockages

SeppeDR, 2023-12-27:
Hi there,

I'm using the Prusa Mini for over 2 and a half years, but recently for mass production of parts, about 200 a day. This uses a lot of filament. As there is no automatic filament switching when the filament is finished, I often awake in the morning to the message that the heater has been disabled after 30 minutes of inactivity.

This is a great feature, however, in those 30 minutes, my pla filament gets cooked in the nozzle. I have 4 high flow nozzles that have become completely useless as they are extremely difficult to clean and cold/warm pulls do not work on them. Needling is the only solution, however, risks of ruining the internal structure of the nozzle. That's about a 100 euros in the trash by now.

I wonder whether the 30 minutes of time is not excessive? It does not take long for the heater to reheat the nozzle to the right temperature. And if it would be switched off after a few minutes, I'm sure the pla wouldn't be cooked.

Just letting it know, and hoping this could be perhaps changed if future firmware, or a setting for the user to choose. Of course I know that there is some excellent solutions on my end as I could take more care to replenish the filament before it runs out, but that would mean a lot of spools with a little bit of filaments left or just a lot more work in general to make sure the nozzle is not getting blocked.

I had bad batches for over 2 weeks, with problems worsening. Took me a while to figure everything out. Could also help with less people having to contact Prusa support for clogged nozzles.

Anyways, thanks for reading. xx

3d-gussner, 2023-12-27:
I am not sure if [`M86`](https://reprap.org/wiki/G-code#M86:_Set_Safety_Timeout)  is active on the Mini but you can try it to add this to your gcode files and PrusaSlicer profiles.
It has to be added to each gcode file as after reset or power on the default values are active again.

SeppeDR, 2023-12-27:
I searched into the M86 code, but it says it will disable the heating of both hotend and the bed, and the bed should not cooldown, as it will make the print come loose. Perhaps there is another code?

3d-gussner, 2023-12-27:
@SeppeDR Please describe the issue in steps as I don't understand why this happens.
I don't own a MINI so can't tell how it should behave or not.
Example:
1. Start print
2. Filament runs out
3. Filament sensor triggers
4. Filament is unloaded
5. Bed stay heated
6. Hotend stays heated.

dstulken, 2024-01-03:
> @SeppeDR Please describe the issue in steps as I don't understand why this happens.

Presumably it's this:
0. Equip printer with Bondtech CHT nozzle or equivalent, with small internal passages
1. Start print
2. Filament runs out
3. Filament sensor triggers
4. Filament is unloaded
 - Some small amount of filament remains in the nozzle after the filament unload.
5. Bed stay heated
6. Hotend stays heated
7. Time passes
 - the filament which remains in the nozzle is baked/ruined during this time
8. The hotend turns off automatically after 30 minutes

Presumably the solution is to either change the safety timeout from 30 minutes (agreed that this is excessive) to something short (5 minutes proposed above)... or to insert a new step between 6 and 7, which drops the nozzle to an intermediate "keep warm" holding temp after 5 mins or so, to avoid baking the filament in the time remaining until the safety timeout elapses.

tysonsw, 2024-01-04:
This probably has to do with the same EU law that states that coffee and espresso machines with a heather must be automatically turned off after 40 minutes of standby. 
It is a safety thing.

SeppeDR, 2024-01-15:
I think that's a very well structured note on my issue. thanks for that (; 👍 
I think choosing the time could be great here. In any case the bed shouldn't stop heating of course, or the print would come off, only the nozzle should go down in temperature.

> Presumably it's this:
>
> 0. Equip printer with Bondtech CHT nozzle or equivalent, with small internal passages
> 1. Start print
> 2. Filament runs out
> 3. Filament sensor triggers
> 4. Filament is unloaded [...]