prusa3d/Prusa-Firmware-Buddy #3643: [FEATURE REQUEST] PHA filament support

pbozeman, 2023-12-28:
Both ColorFabb and BeyondPlatics produce a PHA based filament that is more compostible than PLA and PLA based filament blends.  PHA can be broken down in a backyard composter, v.s. needing special industrial composters like PLA.  PHA also contains no microplastics.

I almost print exclusively in PHA due to to being more environmentally friendly.

There's already an open pull request for PHA support: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2210, but since it is so old, it isn't clear that old pull requests are being triaged.  This feature request is an attempt to bump this for your attention.  It would be great if you could either land the request, or do something to support PHA on your side. 

Thanks - because I'm sure you all care about the planet too ;)

bkerler, 2023-12-31:
Right now mainly the profiles are added that are commonly used. But there has already been a discussion to add an option for custom filament type slots instead.

bkerler, 2024-01-02:
I added a new PR which allows usage of custom filament profiles, which is even more convenient :)

TinkerNinja, 2024-02-16:
+1 for this - have some 100 hours experience with PHA printing if you need info/support/beta testing

closed 2024-07-01 by github-actions[bot]