prusa3d/Prusa-Firmware-Buddy #3646: [BUG] Cancel object menu in MK4 supports only first 16

mixer3d, 2023-12-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI, MK4, XL]
MK4

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.2

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB  

**Describe the bug**
Cancel object menu supports only first 16 instances, even when print plate has few more.

**How to reproduce**
Put more than 16 objects on the plate and try to cancel one of them (over 16).

**Expected behavior**
 I would like to be able to cancel any of the object from the plate, not only one from the first 16. Which is handy sometimes while printing more smaller details...

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

bkerler, 2024-01-06:
That's a firmware limitation as currently, only a max of 16 menu entries can be shown.

mixer3d, 2024-01-12:
ah, ok thank you for letting me know @bkerler

rcmaniac25, 2024-01-14:
Yea. I hit the same issue on the XL and MK4 #3418

bkerler, 2024-01-21:
As the firmware is for MK4/MK3.9/XL/Mini, the current limitation of 16 affects them all.

tjstansell, 2024-03-01:
FWIW, I have a print with 58 objects on it and I was unable to use `cancel current` on any instances over 16.  It beeps when I hit the button (while it's printing mid-air on an object no longer on the bed), but doesn't ever cancel anything.  This leads to a blob getting dragged onto a neighbor and led to several others getting bumped enough to pop off as well.

arekm, 2024-03-01:
16 in one menu? So is it possible to solve this by adding paging and "Next", "Previous" entries?

wafflecart, 2024-04-03:
I know it doesn’t make any difference to the 16 limit but in your slicer if you name the objects before slicing and exporting the names will show on screen. Eg could use object1, object2 etc starting from top left or something. Makes it easier to identify them until they improve the cancel objects menu with pictures or something on build plate.

InnesPort, 2024-05-16:
Came here looking for a solution since this is still happening to me on latest MK4 firmware (6.0.0+14794) and PrusaSlicer (2.7.4).

I (foolishly, I suppose) am currently trying to print 250 small parts and 2 of them have failed.  I've tried using "Cancel current" on the MK4 itself and using cancel object within PrusaConnect since it identifies and shows the map of all 250 instances, which is really cool if cancelling actually worked.

Nothing works.  I simply have to either cancel the print and start over or babysit the print, pause it when it prints in mid air over the failed objects, clean off that bit of filament, then resume print. 

**_Please Prusa_, let us have more than 16 cancelable objects!!**
[image]

mixer3d, 2024-05-16:
> I know it doesn’t make any difference to the 16 limit but in your slicer if you name the objects before slicing and exporting the names will show on screen. Eg could use object1, object2 etc [...]

but then the objects on the plate are in the random order, and you have to print anyway to see and note, which one is in which position

mixer3d, 2024-05-16:
> Came here looking for a solution since this is still happening to me on latest MK4 firmware (6.0.0+14794) and PrusaSlicer (2.7.4).
> 
> I (foolishly, I suppose) am currently trying to print 250 small parts and 2 of them have failed. I've tried using "Cancel current" on the MK4 itself and using cancel object within PrusaConnect since it identifies and shows the map of all 250 instances, which is really cool if cancelling actually worked.
> 
> Nothing works. I simply have to either cancel the print and start over or babysit the print, pause it when it prints in mid air over the failed objects, clean off that bit of filament, then resume print.
> 
> **_Please Prusa_, let us have more than 16 cancelable objects!!** ![IMG_2123(1)](https://private-user-images.githubusercontent.com/17229243/331364115-b068765c-99f4-4066-898b-39d06316daac.JPEG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU4OTI4NjQsIm5iZiI6MTcxNTg5MjU2NCwicGF0aCI6Ii8xNzIyOTI0My8zMzEzNjQxMTUtYjA2ODc2NWMtOTlmNC00MDY2LTg5OGItMzlkMDYzMTZkYWFjLkpQRUc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxNlQyMDQ5MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05NDA2ZGNmZmNlNjMwODk5YWJkMGFhYTdjZTdiODQ1ODQ0MTVkMWExOTFmZjliNzIwODEyOGY4OTI3NDc3NWE0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.gJ4DHA0fZMYjCV1lSCICb_DNTZZuty2MOAIpBExvPoY)

i'm not sure if this can be technically possible, you see already, how slow the menu is during scrolling... 
if you print with petg, just increase the temp. bed to 100 degrees, that way it will reach some 85+/-5 and the petg will stick good enough, and important to use latest config with changed value of the Zlift...

mixer3d, 2024-05-16:
but of course if the bed adhesion is not the particular problem, this will not solve the problem

InnesPort, 2024-05-16:
> but of course if the bed adhesion is not the particular problem, this will not solve the problem

I don't _think_ adhesion was the issue on this print.  All remaining 248 objects are sticking great, I recently washed the sheet and wiped it down with IPA just before this (my usual habit), and the Zoffset looked good on this one and other larger prints.  It just seems like it wigged out for one of those, and then my guess is it took the next one in line out.  But regardless of the problem being able to cancel them would be great... 

Here's a screenshot of the PrusaConnect job info screen where you can easily identify which one is which without having to rename the objects.  I can even highlight any object and hit cancel, then a message pops up that says "command to cancel object has been successfully sent", but then nothing actually changes.

[image]

rkost, 2024-07-31:
The limit is set in these two files - one belonging to the prusa connect functionality:

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/ce6d58b80c995dd8d5923d39688b195a6ac10b1e/src/common/marlin_vars.hpp#L340

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/ce6d58b80c995dd8d5923d39688b195a6ac10b1e/src/connect/printer.hpp#L60

Can someone elaborate what the root issue here is? Is it a memory or performance concern with older printer models? I cannot see any comment either in the source code or in one of the linked issues about the reasoning to set the number to 16.

bgyarfas, 2024-09-16:
It appears you cancel up to object 64 in PrusaConnect. Still not enough...

nomike, 2024-11-14:
> but then the objects on the plate are in the random order, and you have to print anyway to see and note, which one is in which position

What I did is to select the objects in the slicer starting from the top left and naming them like in a spreadsheet. So, first row is A1, A2, A3, A4, etc., second row is B1, B2, B3, and so on. In the example @InnesPort posted above, the first one could either be A1 or D1. That's ultimately up for you to decide as it's you who needs to be able to match these labels to the objects at the end of the day.

It is a tedious task however and now that I've learned that you can do this visually using Prusa Connect, I will rather do that.

This spreadsheet naming approach is good though, if you don't want to or can't use Prusa C
onnect.

CZDanol, 2024-12-13:
Internal ticket: BFW-6465

niobos, 2025-01-18:
I've hit this limitation as well.

If the limitation is actually the menu system, I think [premiumcola's idea](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3646#issuecomment-2035544774) sounds really good: show the previous 8 and the next 8 objects that have just printed / will be printed. In practice, I only cancel objects that "just went bad", or that the tool head is about to hit.

Of course, if the limitation is somewhere deeper in the system, the above solution won't work.

CZDanol, 2025-01-19:
Showing a few previously printed objects is a neat idea!

Showing the "next 8 objects" will probably not be easily doable though, because that would require forward-scanning the gcode.

oriddlero, 2025-01-23:
> Showing a few previously printed objects is a neat idea!
> 
> Showing the "next 8 objects" will probably not be easily doable though, because that would require forward-scanning the gcode.

Have you made any progress on getting this to work through the prusa connect Dashboard cancellation image?  I also tried cancel current directly from the printer on the failed object and it did not register/would not work.  I'm really suffering over here.   lost an entire bed on a three day print  yesterday because I couldn't cancel an object. 

[image]

oriddlero, 2025-02-01:
Any updates? Anyone??  Please???

CZDanol, 2025-02-02:
Sorry, no updates at this moment.

Tupson444, 2025-02-02:
@oriddlero In the meantime you can use a workaround: merge groups of objects in the slicer (so that there are up to 16 groups), so if one detaches you can cancel the group (instead of the whole print). This is suboptimal workaround obviously, but better than nothing.

CZDanol, 2025-03-06:
Hello, just merged support for cancelling 1024 objects. If all goes well, it should come out in 6.4, but I don't have any estimates about the release date.

CZDanol, 2025-03-06:
I suppose it will be like with every other release - first an alpha, then possibly a few more pre-releases, based on what problems will get reported by the community, then a proper release once we are confident enough in the overall stability.

We currently don't have any plans on doing feature previews specifically, sorry.

oriddlero, 2025-03-20:
XL's printing 370+ objects per plate, each print is three days.  

[image]

3d-gussner, 2025-10-15:
@mixer3d Thanks for answering the question from @rcmaniac25 so fast.

Please consider to close this issue.