prusa3d/Prusa-Firmware-Buddy #3647: [BUG] Prusa XL reports incorrect filament type loaded into tool

OneEyeLeft, 2023-12-29:
**Printer type** -XL
Printer firmware version - 5.2.1

**SD card or USB/Octoprint**
USB

**Describe the bug**
When starting new prints on a 5 tool XL that have been loaded to the USB drive via Prusa Link the filament type for some tool heads seems to change without user interaction or a recent filament change. Example: I have the following set up right now:
Tool1 - .4mm nozzle - PLA
Tool2 - .4mm nozzle - PLA
Tool3 - .4mm nozzle - PETG
Tool4 - .4mm nozzle - PLA
Tool5 - .6mm nozzle - PLA
In Prusa Slicer I slice my model for Tool1, .4mm, PLA. When I transfer the bgcode to the printer and select it the printer reports that the current filament type for Tool 1 is PETG, not PLA. I have not changed filaments for a while and have printed with several tools, including tool 1 over the last few days. I have to select Tool 1 and choose to accept the warning that "Tool 1 has the incorrect filament".

**To Reproduce**
I do not have steps to reproduce but I realized this is the second time this has happened to me. I do print .6mm, .4mm sliced objects and do print PLA, PETG, and TPU. The only action I think that was possibly related was a filament change on Tool 5 from PETG to PLA a day ago but I have since printed on Tool 5 and Tool 3 with no issue.

**Expected behavior**
Prusa XL will retain knowledge of its loaded filament type between prints and jobs, especially when the filament has not changed.

**G-code**
No gcode relevant to this bug

Please note in the photo attached that only tool 3 had PETG loaded when this error occurred.
[image]

peter12311, 2024-01-01:
I have the same issue trying to print PLA supports for PETG object. Causes also reboots or red screen on the printer. I couldn't figure out any failure on my side, seems very much to me that whatever piece of Software is messing things up. Using latest Firmware and PrusaSlicer with 5TXL. Why has this one been closed?

OneEyeLeft, 2024-01-02:
Seconding the questioning of the close. No response, no suggestion, no request for specific feedback. What's up Prusa team?

kvtorp, 2024-01-21:
The same has happened to me twice now. Both times it has been on tool 1, and the unexpected filament type was one that I had loaded on a different head - first time, flex instead of PETG (loaded on tool 5), second time PLA instead of PETG (loaded on tools 2 and 3). I upload my files through Prusa Connect

peter12311, 2024-04-05:
I found out that "old" filament profiles in PrusaSlicer led to this error. Recreating the Printer- and Filament profiles fixed this for me.

closed 2024-07-01 by github-actions[bot]