prusa3d/Prusa-Firmware-Buddy #3649: [QUESTION] GPIO "expansion?" pinout addressable via G-code (M42 & M226)?

Fil4ment, 2023-12-31:
I am wanting to use [M42](https://reprap.org/wiki/G-code#M42:_Switch_I.2FO_pin) & [M226](https://reprap.org/wiki/G-code#M226:_Wait_for_pin_state) as outputs & inputs to set and wait for enclosure temperature, coming from an arduino-controlled circulation system.

I only need to know the pin locations that are available to use ("expansion" header on the board looks viable, but I don't have a schematic), and what their pin numbers would be, in the firmware. Can that be shared, yet? If not, is there an ETA?

I've posted this question on the Prusa Forum, [here](https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/xbuddy-board-gpio/) but I've hit a bit of a wall. Any information would be appreciated!

Zappes, 2024-01-01:
I would also love to get an answer to that question. I have some ideas for enclosure mods that could really benefit from such a communication mechanism.

bkerler, 2024-01-04:
Actually that question isn't fw but hardware related, so wrong place to ask (maybe ask the Prusa Support).
Regarding the expansion port, this is what I have reversed so far (Board rev 0.2.7 2021 MK4):

```
    - .. ?
    ? .. ?
  PE5 .. PE3
  PE4 .. PE2
 PA15 .. PF8
    - .. -
    + .. +
  EXPANSION 1
```
The PIN definition can be found here:
[hwio](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/common/hwio_pindef.h)

Fast IO numbers can be found here:
[io_pin](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/lib/Marlin/Marlin/src/HAL/HAL_STM32F1/fastio.h)
PE3 = 0x43
PE4 = 0x44
PE5 = 0x45
PF8 = 0x58
PA15 = 0x0F

Note: This is what I have reversed, so no guarantee for that. However I don't think that M42 or M226 can be used for that.

Fil4ment, 2024-01-05:
I actually asked Prusa Support first, and they told me to ask the devs. Since I'm asking for the *addressable* pin numbers, I think that qualifies as a firmware question. Thank you for helping with the pinout. That is hopefully a step forward.

bkerler, 2024-01-13:
As the schematics have been published [here](https://www.prusa3d.com/page/open-source-at-prusa-research_236812/), it's now pretty clear.

Fil4ment, 2024-01-13:
Yup. Thank you for your help in this, and documenting it so clearly.

closed 2024-01-13 by Fil4ment

Hilfarnal, 2024-07-03:
> Yup. Thank you for your help in this, and documenting it so clearly.

Can I please ask you to say a little bit more about how did you achieve this? I'm looking for working GCoode used as signal output and Pin number from Expansion port

Fil4ment, 2024-07-04:
There isn't any. The M43 command is supported, but the "analogwrite" function is restricted to only a few pins, and it doesn't seem like they're going to change that, any time soon. If you want to output or input anything, via gcode, [THIS](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3840) got merged with the code base, and should be included in some future release.

Hilfarnal, 2024-07-04:
I've asked the question before diving in deep into the topic and getting myself up to date with all the open tickets. Thank you for the response. Although using i2c unnecessarily complicates the development of simple peripherals it will still be a nice addition .

Hilfarnal, 2024-08-12:
I guess after today's release of GPIO hackerboard the topic is live again. Although MINI is not officially supported yet the fact that it's based on i2c and new MINI firmware will give us access to it I'm fairly sure we will be able to connect it after changing cable connector.  Any official hint that MINI owners should also preorder the board?

mishak87, 2024-08-14:
@Hilfarnal I checked the schematics and you can definitely splice the cable and plug it in the expansion connector but question if/when it will be enabled in the firmware for Mini.
The board costs 2-5x times more than Attiny8X/Arduino/ESP32 dev board that you can connect as I2C slave and some have even Wifi/BLE...

It will be a hard sell for DYIers since you have to solder everything to it anyway or use 4 pole jack. You might as well order connector 5025780400 and few pins and make wire to the dev board yourself...