prusa3d/Prusa-Firmware-Buddy #3650: [BUG] Missing outer perimeter.

Mikyner, 2023-12-31:
Occasionally, the outer perimeter of a print is missing.
[image]
[image]

MichaelJLew, 2024-01-04:
Is the perimeter in the gcode? If not then this is a slicer issue rather than a firmware problem. 

I will note that you have not given any of the information requested by the issue report form. Perhaps you are venting rather than reporting an issue that can be fixed by Prusa...

tomoslono, 2024-01-06:
I'm having this same issue on PrusaSlicer 2.7.1.

the-keybord, 2024-01-10:
I have the same issue, it happens random , on different models, different settings, once i saw it happening with my eyes, printer just skipped a few moves of perimeter and then continued printing as normal , 
After that i put the same gcode on print, and issue did not happen,

I contacted prusa , the guy told me there maybe be a faulthy wifi connection or usb drive, but gave me no solution for this

It looks like the printer just skips a few gcode lines and forgets about it

the-keybord, 2024-01-10:
slicer is good because the same gcode worked the second time, maybe its because i use prusa connect?

steinack, 2024-01-10:
I have experienced similar problems. Since I disabled "Use ramping lift" in PrusaSlicer, I no longer have any problems. 
[image]

the-keybord, 2024-01-10:
I see you had the exact issue i have, but i dont even have "use ramping lift" enabled

tomoslono, 2024-01-11:
For me the issue resolved after switching from Arachne to Classic generator in Slicer.

SmithyAT, 2024-03-03:
Had the same issue today twice. Different models, different printers.
I have two MK4 printers and printed a model on the first one and saw a missing outer perimeter about 10mm above the bottom. The perimeter is missing in the whole layer-
Later I printed something else on the other MK4 and had the same issue, but not on the same height, on the second print it was maybe 12mm above the bottom.
It don't happen always. Printed some other things today and they were fine. 
I updated to 2.7.2 yesterday and haven't seen the issue before, so I thought 2.7.2 is maybe causing it. Then found this bug here, so I don't think it is related to 2.7.2.

the-keybord, 2024-03-03:
do you use usb drive directly or Prusa Connect ?or Prusa Link

the-keybord, 2024-03-03:
@SmithyAT , provide some info about your settings, are you uploading directly on usb drive, or use wireless? Prusa link or Prusa Connect?

SmithyAT, 2024-03-03:
I am using Prusa Connect.
Do you think it is related? I always thought that there should be no difference, because the gcode is stored on the use drive anyway.

the-keybord, 2024-03-03:
@SmithyAT yes, because i use Prusa Connect too,  but some times i used usb drive, and issue never showed, i think there is a probability, we should switch to usb drive and see if this issue ever shows up, 
because i used the same gcode , and first time issue persist and second time no,

SmithyAT, 2024-03-03:
> @SmithyAT , provide some info about your settings, are you uploading directly on usb drive, or use wireless? Prusa link or Prusa Connect?

I am using default settings in prusa slicer, the model is quite simple, a block with a few cut-outs on the top (it is a holder for some things in the bathroom).
But I've seen your suggestion to turn off ramping lift, and checked my settings, it was enabled, seems to be on per default. Will try to disable it with the next print.

> Do you use upload, or upload and print ?
Both, but as far as I can remember I clicked "upload and print"

But good hint, will try to copy it manually to the usb drive. WIth the wifi speed and the general transfer, everthing could happen :-) Maybe there is no check if the checksum is correct after the transfer. 

But as already said in my initial posting, I've never seen the problem before and I never copied the gcode manually, I am always using PrusaConnect. But it is worth to try it, thanks!

the-keybord, 2024-03-03:
@SmithyAT i had these issue pretty often , but on very random time, i was using "upload and print" button, but now for a few months i click "upload" and then wait for the file to full upload, and then i click "print" on the printer, and that way issue never showed up, but im just really venting about this nonsense bug , im sorry xd

SmithyAT, 2024-03-03:
I can understand that, it's annoying.
Will take care next time, how I send the job and will give it a try to just upload it. 
Hope it will be fixed soon, these workarounds needed for PrusaConnect are really annoying, its not the first one, since I used it.

hstag, 2024-05-05:
I have same issue on my mk4, firmware 6.0.0, Prusa slicer 2.7.4. I am using prusa connect.

Koder22, 2024-05-12:
I have the same issue with my Mk4 and mini, when using wifi there is a random missing/extra line in the prints.

[image]
[image]
[image]

Koder22, 2024-05-19:
Here is some more examples

[image]
[image]
[image]

antimix, 2024-07-11:
I never experience this, but because I always send the file via PrusaConnect and then I print from the LCD panel.
This can be related to the WiFi BUG that afflicts the MK4 since it was born (see: [ [BUG] Prusa Connect / Link Gcode Corruption on OTA gcode upload #3156 ](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156)

Do the following:

1. Configure in PrusaSlicer a physical MK4 printer with PrusaConnect.
2. Slice the model in PrusaSlicer and export on your PC.
3. Then upload the same code to the printer using the G button.
4. Wait for the complete transfer.
5. Go on the LCD and exit from the ready to print screen and back to the main menu.
6. Remove the USB key and connect it to the PC
7. Binary Compare the two files if are the same.
   here is a quick and dirty way: 
   - go in windows command line with Start and select Command Prompt
```
   C:\Users\john>fc /b PCobjectName.bgcode D:\MK4objectName.bgcode
   Comparing files PCobjectName.bgcode and D:\MK4objectName.bgcode
**FC: no differences encountered**
```

If you get the "FC: no differences encountered" then the file was not currupted during PrusaConnect transnsfer.
8. Reinstall the USB dongle on the MK4

9. Go on the LCD panel, select the object that you have checked and print it.
10. I suppose that 100% of time it will be perfect. No way to fail.

11. Then go in PrusaConnect, and you should be able to see the same object also there.
12. Print it from PrusaConnect
13. I am not sure if PrusaConnect is just printing the local USB copy or is **resending** the whole file.
14. If it is using the local copy the it should be 100% correct.

15. If you get a bad print with a line, then may be it resent the gcode. In this case after the print, remove the USB. connect it to the PC, and reissue the FC command from dos command. Then the bgcode file should be different.

Let us know.

Regards

crasbe, 2024-08-01:
@antimix I experienced issues with my new Mk4 as well. In particular, at a certain height it moved veeeeeeeeeery slowly (barely noticeable). But it recovered after I paused the print (which took forever as well, probably 5 minutes or more).
Unfortunately I did not follow your guideline because I only researched the issue after it happened.

The printer has the latest 6.0.4 firmware.

This is what I know:
- The file was transfered to the Mk4 via PrusaConnect from PrusaSlicer 2.8.0 and started directly. The issue occured at a height of 16.65mm on a move without extrusion and continued during the next movements with extrusion.

- It seems to have recovered after pausing the print, but the pause took very long to be in effect.

- After the print finished (well, it failed, because the part came off the bed. Not sure if that's related or not.), I downloaded the .bgcode file and it is 100% identical with a file that was saved locally.

- Then I disabled PrusaConnect and PrusaLink (I did not use PrusaLink, it was just enabled) and restarted the same file and it did not have the same issue. However the printer was powercycled in between.

- Previously I had issues with incomplete outer perimeters as well, which seems to be gone now too.

- The issue has been present in two different WiFi networks in two different parts of Germany with different internet connection speeds (50MBit and 16MBit). Both routers are Fritz!Box routers, however different types.

michael-kobb, 2024-09-03:
I had this issue the other day on one of my first prints from Prusa Connect to my MK4. The printer simply skipped some of the gcode on the top layer, leaving a sort of diagonal stripe of missing material. I sent the print directly from Prusa Slicer 2.8.0 to my MK4 which is running 6.0.4+14924.

I downloaded the gcode from the printer and exported it from the slicer. Comparing the two files using the diff utility on the Mac, they are identical.

[image]
[image]

michael-kobb, 2024-09-04:
Phooey. Happened again, this time on my XL. Apologies for the quality of the photo. I snapped the pic while it was still printing and it was ASA so the enclosure was closed. This was on my XL 5T running 6.1.2.
[image]

Dorito203, 2024-09-07:
I have the same problem (Firmware 6.1.2+7894) and the frequency has increased in the last few weeks.
I now have this issue in almost every print, in the beginning it only occurred quite rarely.
I transfer the files via Prusa Connect and usually start the print from the display.

Koder22, 2024-09-07:
> I had this issue the other day on one of my first prints from Prusa Connect to my MK4. The printer simply skipped some of the gcode on the top layer, leaving a sort of diagonal stripe of missing material. I sent the print directly from Prusa Slicer 2.8.0 to my MK4 which is running 6.0.4+14924.
> 
> I downloaded the gcode from the printer and exported it from the slicer. Comparing the two files using the diff utility on the Mac, they are identical. And of course if I look at the gcode file with the gcode viewer utility in PrusaSlicer, it does not show any part missing.
> 
> ![IMG_3561](https://private-user-images.githubusercontent.com/180352937/364169873-7daabc66-bf88-4e18-8159-877c96d91d15.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3MjMwMzAsIm5iZiI6MTcyNTcyMjczMCwicGF0aCI6Ii8xODAzNTI5MzcvMzY0MTY5ODczLTdkYWFiYzY2LWJmODgtNGUxOC04MTU5LTg3N2M5NmQ5MWQxNS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MDdUMTUyNTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzhhM2RlOWZhMDcyY2QyNmMwMDg2ZmQ5M2Q2MjhiYzgwM2QxNzU0Mzg5MWU0YzQ5YzY0YzIwYmFjZmJjYmNjZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ED7bgCI1FCbZGBnYL_E2lo_U2YJ7U9m5qamLSxCwOZ0) ![IMG_3560](https://private-user-images.githubusercontent.com/180352937/364169875-a1ab9471-84a6-438d-972a-57f2f19482d8.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3MjMwMzAsIm5iZiI6MTcyNTcyMjczMCwicGF0aCI6Ii8xODAzNTI5MzcvMzY0MTY5ODc1LWExYWI5NDcxLTg0YTYtNDM4ZC05NzJhLTU3ZjJmMTk0ODJkOC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MDdUMTUyNTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWI0NWMwNTJjZWYzNTZkNTk2OThiYWFjOTYxY2IzNDk2ZTY0NDYyNTVmZGVhM2VjNGEzYTY3OWMwYzEyYWJjYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.FvFfof3b1dRoqHOSdVLTMI8vYYSzS-lbPAvpAo0ADXI)

I have had this happen also, a couple of times. Did the same comparison test and found that g-code was identical.

WeaselWasher, 2024-09-20:
Ugh.

Me too.
I also had the XL USB drive disconnected issue. 
Wonder if it's related. 
I sent all of it off to prusa for analysis.
[image]

[image]

[image]

I also had one support on the red print fall over - not sure why - but it's like that one layer just didn't fully print and it fell over.

These are two different iterations on two different versions of firmware and the red one is also sliced with the newest prusa slicer released two days ago.
I transferred the file directly to the USB from my computer. I have prusalink and connect disabled on my XL, though - wifi is still enabled.

CZDanol, 2024-09-22:
@WeaselWasher 
> These are two different iterations on two different versions of firmware (XL 6.1.2 and 6.2.0 alpha) and the red one is also sliced with the newest prusa slicer released two days ago, where the black was the previous version.

I don't understand what you meant by this.

Also, yes, guys, if you are having skipped gcode issues like what I see on the photos from @michael-kobb, please try out 6.2.0, should be fixed there: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases

WeaselWasher, 2024-09-22:
@CZDanol 

By different iterations, I mean that it was the same model, repeated.  

The black print was done in 6.1.2 and the red was done 6.2.

CZDanol, 2024-09-22:
So, you had a USB disconnected issue on 6.2.0?

WeaselWasher, 2024-09-22:
> So, you had a USB disconnected issue on 6.2.0?

I had no fault, but I did have the missing perimeter (this is a thread for missing perimeters) is this the same problem as the USB disconnected issue? 

I have a different petg print going right now that's been going for 12h so far, in two colours. But I haven't noticed any issues yet. It's only half done though. I won't be able to fully inspect it until it's done, but so far no USB disconnect faults or visible missing perimeters

WeaselWasher, 2024-09-22:
@CZDanol 

Yes. It appears I am still getting the missing perimeters in today's print with 6.2
[image]

It's a little hard to see - as it's translucent. But you can definitely see the blob

hstag, 2024-09-22:
Firmware 6.1.3 issue still persist.
[image]

michael-kobb, 2024-09-23:
> Also, yes, guys, if you are having skipped gcode issues like what I see on the photos from @michael-kobb, please try out 6.2.0, should be fixed there: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases

I'd like to be sure that I understand what 6.2.0 contains. Do you believe that the missing perimeter issue that is described here is related to the USB disconnection issue? I have had the missing perimeters happen when using Prusa Connect (never when Prusa Connect is disabled). I have only ever had the USB disconnect problem occur when trying to start a print, never during the print.

If 6.2 is supposed to resolve the missing perimeter issue, I will install it and give it a try, but it will probably be next week before I can do so due to other commitments.

CZDanol, 2024-09-23:
I'm not convinced these missing perimeters are related to the USB - so we're not expecting the perimeters to be solved (it would be very nice though). We have however found out something somewhat related to connect and fixed it, so maybe things could get better as well.

Got an idea - do you have Touch Sig Workaround enabled by any chance?

michael-kobb, 2024-09-23:
> Got an idea - do you have Touch Sig Workaround enabled by any chance?

I do not have that enabled. (I had to google it to find out what it is!) My touch screen seems to work fine.

I agree it would be very nice if this were fixed, but it's a little worrying that this issue, which is now nine months old, does not seem to be under more urgent investigation. It renders Connect unusable on both of my printers.

michael-kobb, 2024-09-24:
Connect doesn't work through wi-fi?? Man, I'm so glad you said that. I was just about to update the printer, but that would have done me no good at all. I don't have Ethernet at that location.

WeaselWasher, 2024-09-24:
Yeah. I'm not sure what happened. It's reported already by others. It just denies the connection. Could be a troubleshooting step on purpose. It is alpha after all. I strapped mine in with Ethernet somewhat temporarily. Wanted to try using connect again since danol said they found and fixed an issue with it. 
I also just replaced my ribbon cable with a double shielded cable, with the shield grounded on the main board side. I've got another 15h run going. So I guess we'll see how my perimeters look tomorrow.

michael-kobb, 2024-09-24:
Good luck! I don't have Ethernet where the printers are, so it would be a challenge to try out 6.2 with Connect. Maybe that will be fixed before I get back from my travels.

WeaselWasher, 2024-09-25:
> I also just replaced my ribbon cable with a double shielded cable, with the shield grounded on the main board side. I've got another 15h run going. So I guess we'll see how my perimeters look tomorrow.
> 

Well. It finished, and I can't see any missed perimeters on this one. 
[image]

crasbe, 2024-09-25:
> Also, yes, guys, if you are having skipped gcode issues like what I see on the photos from @michael-kobb, please try out 6.2.0, should be fixed there: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases

So far I printed for about 30 hours on the XL with Prusa Connect via Ethernet and no perimeter (or other) issues arose. However I never tried it on this XL with Prusa Connect before after the issues with the MK4.
I have yet to try it on my MK4 with WiFi, which previously had the perimeter issues.

Koder22, 2024-09-26:
I did the following on my MK4(factory built, no mods), printed 6 benchys with Prusa Connect(half with "upload and print" and the other half with "upload" and then pushed print on the printer. Result: 3 of the 6 benchys had missing perimeters at the height of 15-20mm.

Then i printed 6 benchys using Prusa Link, and all came out nicely without any missing perimeters.

I cant find any difference between the file thats transfered to the printer, and the one on the computer with the slicer. The missing lines doesnt appear on the same place with the same GCODE. 

[image]

CZDanol, 2024-09-26:
@Koder22 that was a very nice engineering approach. Thank you for that :)

Just to clarify some things:
- Those 3 out of 6 benchies, I assume those were the benchies that you printed using "upload and print"?
- What was the version of the firmware you were trying this on?

Koder22, 2024-09-26:
I printed 3 benchys with "upload and print", two of these had missing perimeters. Then i printed 3 benchys with "upload" and then "print" on the printer, one of these had missing perimeters. I have Prusa 6.1.3 on my printer, i have had these issues since december 2023 when i got the printer.

Koder22, 2024-09-26:
The upper Benchy is printed with Prusa Connect, but didnt have any missing perimeters. The lower one is printed with USB transfer. There is a lot of small defects in the upper one, and these are also present on ones printed with Prusalink.

[image]

the-keybord, 2024-09-27:
@Koder22 Thanks for your investigation
I also want to clarify something:
The prints when you hit "upload" and then "print" , did you wait for the file to load 100%? or you just click print the same moment it displayed on the printer?

My way to avoid this issue is i hit "upload" , then i wait for the whole file to load to printer, and then I click print, 
for a couple of months of using this technique i never had issues with skipped gcode or perimeters

Thanks

crasbe, 2024-09-27:
> So far I printed for about 30 hours on the XL with Prusa Connect via Ethernet and no perimeter (or other) issues arose. However I never tried it on this XL with Prusa Connect before after the issues [...]

So far the MK4 printed close to 20 hours without a single issue, always using Prusa Connect to upload the file and immediately start the print afterwards. The XL has not shown any perimeter issues either after another 20 or so hours.

<hr/>

@Koder22 I don't think this is an EMC issue, considering 6.2.0-alpha1 fixes the issue for me.

From a physical standpoint, the stepper cables have a much higher potential to radiate EM fields than to be affected by EM fields. WiFi works in the Milliwatt range while the steppers are driven with much more power (that's why they get warm).
Apart from that both work in different frequency ranges, and a big electromechanical device such as a stepper wouldn't show much to any response to high frequency signals such as WiFi.

The stepper control is open loop, so there is no feedback from the stepper to the controller that could somehow be affected either.

Racemade, 2024-09-28:
Have the same issue on my XL. One outer perimeter is missing.[image]

michael-kobb, 2024-09-28:
Was that printed from Connect?

Racemade, 2024-09-28:
Directly over the slicer

Koder22, 2024-09-28:
I installed newest firmware on my MK4, 6.2.0 Alpha 1, and did a testprint. This actually looks really good, it has no dimples/artefacts, and no missing lines. I will print a few more just to be on the safe side, but im hoping that this firmware fixed the error!

[image]

Racemade, 2024-09-29:
Another idea:
Could the slicer option "arch firting" (enabled g2/g3) coase the  problem? I remember that ive checked the box a few days before...before it was unchecked.

MichaelJLew, 2024-09-29:
Enabling G2 and G3 may cause a problem. I previously had a bug report here on G2 and G3 curves having blobs because of short pauses during their execution. That report timed out and has been removed, I think, but the problem seems to have persisted. I have very few arcs (G2 and G3) commands in my print files nowadays as a workaround for the original problem, but I do heat minor stuttering of my MK4 where they are used.

Racemade, 2024-09-29:
Thank you for your comment. I will uncheck the box again. Hoping for 6.2 that the issue may resolved for prusa connect.

michael-kobb, 2024-10-05:
I've been using my XL for the past week with the 6.2 alpha 1 firmware. This release breaks Wi-Fi, annoyingly, so I've been using Ethernet, but I have not seen the missing perimeter issue during this time.

Now Prusa has released 6.2 alpha 2, which fixes the Wi-Fi breakage. But it seems that all they've done for that fix is to revert the prior change, so I fear that it will reintroduce the missing perimeter bug.

helder112, 2024-11-25:
Hello All,
I would like to ask if any  bug correction was done on this matter. I also have this issue on my mk3.5 printer.

[image]

michael-kobb, 2024-11-25:
I've held off updating to 6.2 alpha 2 out of concern that it will bring the perimeter issue back if Prusa Connect is enabled. I was planning that I would maybe update to alpha 3 or whatever came next, but so far there have been no further updates.

I'm very disappointed in how slowly Prusa seems to be moving to address this issue.

crasbe, 2024-11-25:
The Alpha 2 did not bring the perimeter issue back. I've been printing with it for weeks with dozens of days of printing days now.

Racemade, 2024-11-25:
Jep, works fine! :)

WeaselWasher, 2024-11-25:
I'm also using prusa connect via Ethernet on alpha 1 (haven't bothered with
alpha 2 since I'm hardwired) no issues

michael-kobb, 2024-11-25:
That's encouraging, thanks. I'll update to alpha 2 and give it a try.

Koder22, 2024-11-25:
Alpha 2 works on my MK4, no missing perimeters. I also feel that Prusa could have done a lot better on this, i spent much time trying to fix my MK4 since i got it a year ago, and there is still not a working firmware(only a alpha).

[2 more replies]

closed 2024-11-30 by Prusa-Support