prusa3d/Prusa-Firmware-Buddy #3651: [BUG] Ugly seams

Mikyner, 2023-12-31:
The seams look terrible and are very visible. On one of today's copies there is a gap instead of a seam (marked with an arrow on the photo)

[image]
[image]
[image]

sirgrej, 2024-01-02:
I can confirm this issue on mk4 with fw 5.1.2 and prusaslicer 2.7.1. 
first I thought it is an input shaper issue but I looks the same without input shaping 
[image]

MichaelJLew, 2024-01-04:
Seams are likely more to do with the slicer than  with the firmware, so maybe have a look at the PrusaSlicer issues on GitHub.

I will draw your attention to two requests for the same feature: seams that can be nearly invisible. https://github.com/prusa3d/PrusaSlicer/issues/11621
https://github.com/prusa3d/PrusaSlicer/issues/11948

You might like to give the post-processor script that is linked in the second of those posts a go. In any case you might comment on those pages to give them a bump.

FutureSharks, 2024-01-06:
This is a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3286

Maybe just close this and comment there?

Mikyner, 2024-01-06:
I confirm that this is a bug in Prusa Slicer and not in the printer FW

[image]

uncovery, 2024-01-18:
[image]
Same here. It's a disaster. MK3 never had this issue.

Mikyner, 2024-01-18:
> ![20240118_151214](https://private-user-images.githubusercontent.com/583808/297654630-df286224-f627-4990-95b9-98138bd958ca.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1NjYwOTEsIm5iZiI6MTcwNTU2NTc5MSwicGF0aCI6Ii81ODM4MDgvMjk3NjU0NjMwLWRmMjg2MjI0LWY2MjctNDk5MC05NWI5LTk4MTM4YmQ5NThjYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExOFQwODE2MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YWY3MWQ3Mzc4NTZhY2RjNzYyMDdiMzU0NjEzM2YwOTE5ZDMyMGMyNDljYzZjODcxNWY1OGJiYmIxYzhkNDI4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.pT3kPi9gZC2a0JFlriaSg3SZv5-d_TS2bAo9Wgjl6bM) Same here. It's a disaster. MK3 never had this issue.

It's a slicer problem, I'm using Orc

Mikyner, 2024-01-18:
It's a slicer problem, I'm using Orca now and everything is fine.

FutureSharks, 2024-01-18:
I think just close this and refer to https://github.com/prusa3d/PrusaSlicer/issues/11914 ?

bryn51, 2024-03-01:
Exposed and obvious seam lines have been with Prusa printers for a long time.
Now, both Mk4 and XL printers have finer steps it ought to be possible to at least reduce the seam gaps somewhat by way of the higher accuracy of movement.
I have never seen a rational reason given as to why we must tolerate them. 
If prusa saw fit to spend huge resources fixing resonance aberrations with the IS thing, the seam lines ought be worth at least as great an effort since the impact is in fact much greater.
The open seam lines not only look bad, they weaken the print too.

MichaelJLew, 2024-03-01:
Try the latest development build of OrcaSlicer. It has scarf joint seams that fix many seam issues.

bryn51, 2024-03-01:
Thanks for the suggestion.
Buuut…
I’m of an age not to be constantly trying new products, preferring the safe road.

If Orca can fix this issue why not Prusa ?

MichaelJLew, 2024-03-01:
The fix is in Orca, not yet PrusaSlicer. Use it or don’t. It’s not that you are too old.

mixer3d, 2024-03-11:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997

GithubUser99999999, 2024-05-09:
Still not fixed

MichaelJLew, 2024-07-09:
Maybe stale, but still not fixed.

bryn51, 2024-07-09:
Its true.  I do not see why Prusa do not have this as a priority, as the code they need is already developed by others and just needs to be ported over (ok maybe more complex than "just", but still the broad algorithm has been developed already).

> Maybe stale, but still not fixed.

danopernis, 2024-07-09:
Perhaps that's because this issue is open in the wrong repository. This is not a firmware bug. Also, it is a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3286

bryn51, 2024-07-09:
> Perhaps that's because this issue is open in the wrong repository. This is not a firmware bug. Also, it is a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3286

There is a likely disagreement in definitions of what is a bug and what is an enhancement. 

In my understanding one must refer to the product specification (if one exists).
If the product functions according to its specification and you want to alter it, then it's an enhancement.
If the product does not behave according to specification then its a bug.
But unfortunately it does not end there.
If a new version is released and the product is unstable or gives up an error then its a bug in development.
And so it goes on.
Rules differ according to   who you speak to and their role/perspective.
It is totally unreasonable for professional developers to refuse to look at an issue based in an incorrect user classification, and certainly reprehensible to dismiss user reports based solely in misclassification by a user. And yet thats what we see prusa staff doing on this forum.
Whats more reasonable is for a management staffer to look at each report and simply assign it to an appropriate team no matter its original classification (and if necessary reclassify if if that makes sense). 

How can I say this so confidently ? 25 years in IT development, as a developed, contractor, manager.