prusa3d/Prusa-Firmware-Buddy #3653: [ENHANCEMENT] Show M117 (status line/progress messages) on main screen when using serial/Octoprint

coaxial, 2024-01-01:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI v5.1.2]
**Optional upgrades** - N/A

**Is your suggested improvement related to an existing problem? Please describe.**
  I'm always frustrated when Octoprint sends [M117](https://marlinfw.org/docs/gcode/M117.html) status line/progress info to the printer but I have to go dig into the menu to show message history to see it.

**Describe the expected functionality**
  Show the status messages as they arrive on the main screen (instead of the Octoprint logo) so that ETA, progress, etc is visible at a glance while printing with serial/Octoprint.

SetupCoding, 2024-01-22:
We need this.

DiederikvandenB, 2024-02-28:
This would also be nice on the Mk4.

timur-tabi, 2024-04-14:
See also #3846.

timur-tabi, 2024-07-01:
It's only stale because Prusa's developers refuse to fix it.

bkerler, 2025-08-26:
This is the same as issue #189 and should be closed.

jkotowicz123, 2026-02-19:
For real guys, I can imagine you have lots of work, but this is really crucial feature for lots of your power users. I choose Prusa printers over other brands mostly for that access/ tinkering possibility. We rely on our print farm on custom plugins running on Octoprint. As Connect is still nowhere near that advanced, nor open for modification, we still need good support for Octorpint.

timur-tabi, 2026-10-14:
I still want this enhancement implemented.