prusa3d/Prusa-Firmware-Buddy #3655: [BUG] Filament change purge amount for XL and MK4

salmander, 2024-01-02:
**Printer type** - MK4, XL

**Printer firmware version** - v5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive / USB drive + Prusa Connect**

**Describe the bug**
  For filament change, the purge amount is not good enough. I have to at least purge twice more to get the colour changed adequately.

**How to reproduce**
* Filament > Change Filament
* Unload, press ok
* Insert new filament.
* Choose filament type e.g. PLA
* Printer purges some filament and then asks it needs more purging.
* Check if the colour has been transitioned, if not, then purge more.

**Expected behavior**
  Shouldn't need to select "purge more" twice. Just set the default purge amount to be higher.

dstulken, 2024-01-03:
It's the same experience with the Mini+ unfortunately.

Presumably the purge amount is kept low to minimize waste, but having to press "more" so many times is definitely a workflow annoyance.

salmander, 2024-01-03:
I appreciate saving filament but it is counter intuitive if every time I have to select purge more a few times to change the filament.

IftiBashir, 2024-01-04:
I do not always have to select to purge more. Sometimes I do, but a majority of the time the first purge is enough for me......

bkerler, 2024-01-06:
This isn't a bug, but should be labeled "Feature Request" instead.

I assume it depends on the color and the amount of color particles are in the filament. If I switch from white to black, it's sufficient, but switching from black to white, it does take an extra purge.

Not sure how that could be handled better instead... maybe pressing the scroll wheel for purging and releasing the scroll wheel for stopping the purge ?

closed 2024-07-01 by github-actions[bot]