prusa3d/Prusa-Firmware-Buddy #3656: 0.6 Nozzel is set to the Printer - no warning if you send a file with a 0.4 Profile!

beroliv, 2024-01-02:
**Printer type** - [ MK4]

**Printer firmware version** - [5.1.2.]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
 Pusa Link or Prusa Connect

**Describe the bug**
 0.6 Nozzel is set to the Printer - no warning if you send a file with a 0.4 Profile! The opposide way the Printer give a warning the Nozzel diameter are not pass!

**How to reproduce**
  Set the Printer to 0.6 Nozzle send a file with 0.4 Profile

**Expected behavior**
  The Printer should give a warning Message as he do the opposide way!

dpark12300, 2024-01-04:
Are the settings set to warn you if the nozzle is incorrect? Not sure where the setting is though.

beroliv, 2024-01-04:
Yes! 
The opposide way the Printer give a warning the Nozzel diameter are not pass!

dpark12300, 2024-01-04:
Hmm. I don't know what to do.

beroliv, 2024-01-04:
set your printer to 0.6 Nozzle, try to print a file with 0.4 Profile! Did a warning appear? If yes nothing to do! if not something is wrong.

danopernis, 2024-06-06:
This should be fixed since v6.0.0 via commit https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/3840f9a4f388e82eef08d26540d78fab8af6460e

@beroliv can you check if the issue is fixed for you?

closed 2024-08-13 by github-actions[bot]