prusa3d/Prusa-Firmware-Buddy #3658: Přehrání firmwaru prusa mini

Maneekcz, 2024-01-02:
Ahoj dá se přehrát fimwer na prusa mini jinak než z USB. Aktuální verze FW 4.3.4 a dále se nelze dostat. Nechce se přehrát bootloader a aktualizuje se pouze fimvare. Tiskana se dostane do spouštěcí smicky. Když zkouším soubory vkládat postupně také to nepomáhá.

marek26340, 2024-01-03:
A už jste to zkoušel řešit přímo s podporou? GitHub bohužel není helpdesk podpora.
Možná bych ještě zkusil přehrát 4.3.4 - nahrajte na USB, připojte a hned po zapnutí tiskárny mačkejte tlačítko. Měla by se takto spustit přeinstalace. Pak to zkuste znovu.

Ideálně bych ještě zkusil aktualizaci provést s jinou flashkou, kterou na 1000% víte, že je bezchybná, plně funkční, naformátovaná na FAT32 a ne příliš velká (původní co jsem dostal k MINI já má asi 8GB).

Maneekcz, 2024-01-05:
Tak sem vyzkoušel 5 flash disku a žádný posun chtěl bych to zkusit nahrát přímo do desky přes pc jenže to musí být hex

closed 2024-06-27 by danopernis