prusa3d/Prusa-Firmware-Buddy #3660: [BUG] GCode sound playback (M300) is to fast!?

foorschtbar, 2024-01-04:
**Printer type** MK4

**Printer firmware version** 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** n/a

**USB drive or USB/Octoprint** Octoprint

**Describe the bug**
Since a few weeks we got M300 GCode support, but the playback seams to fast

**How to reproduce**
Try this gcodes:

```
;Scooter
M300 S1567 P206
M300 S1567 P206
M300 S1396 P103
M300 S1318 P103
M300 S1396 P413
M300 S1174 P310
M300 S1046 P103
M300 S1174 P206
M300 S1567 P206
M300 S1567 P206
M300 S1396 P206
M300 S1318 P206
M300 S1567 P206
M300 S1567 P206
M300 S1396 P103
M300 S1318 P103
M300 S1396 P413
M300 S1174 P413
M300 S1318 P206
M300 S1046 P206
M300 S1174 P827
M300 S0 P206
```

**Expected behavior**
- In a webplayer https://foorschtbar.github.io/GCode-Ringtone/ -> sounds right (put gcode on the right side)
- On printer (i uploaded a *.gcode file only with that content to octoprint and "print" the file) -> sounds strange

bkerler, 2024-01-13:
Not 100% sure, but I think it's related to this line : [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/dfc2bc6235505eacee24846e1659d693ed040e80/src/common/hwio_XLBuddy.cpp#L404).
```
        per = (uint32_t)(1'000.0F / frq);
        pul = (uint32_t)(del / per);
```
I don't see any relation between del (delay) and the hw clock frequency, but maybe I'm missing something.

bkerler, 2024-01-15:
I just pushed a PR that should fix the issue.

CZDanol, 2024-02-15:
Internal ticket number: BFW-5083

Prusa-Support, 2024-06-22:
The fix was merged at the end of April and made available with the firmware release in May, so that adding a [Dwell code](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1075#issuecomment-1955316303) is no longer needed.

This should close this issue, however, further feedback is surely welcome.

Michele Moramarco
Prusa Research

danopernis, 2024-07-09:
There must have been some misunderstanding. This was merged, but not released. It is scheduled for the next patch version. Sorry for the confusion.

danopernis, 2024-07-19:
New firmware [v6.0.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.4) was just released which should fix this.

closed 2024-08-08 by danopernis