prusa3d/Prusa-Firmware-Buddy #3664: [BUG] Prusa MK4 - Print freezing / stopping / pausing

stuz32, 2024-01-06:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** 
- MK4

**Printer firmware version** 
- 5.1.2

**Original or Custom firmware** 
- Original

**Optional upgrades** 
- None

**USB drive or USB/Octoprint**
  USB drive (bug repeatable on multiple drives freshly formatted in FAT32, so appears not to be related to the drive itself)

**Describe the bug**
- During a print at seemingly random times, the printer will appear to get stuck / frozen / pause
- The fans continue to run
- The screen remains on however the system itself doesn't respond to input via the scroll wheel

**How to reproduce**
- Unfortunately I was unable to reproduce reliably and even using the same simple gcode file, it appeared to not repeat the pausing behaviour at the same position in the print

**Expected behavior**
- The print should continue without pausing unless specified by the slicer to pause during the print

**G-code**
- Not in a position to provide however I have recorded a video of the issue

**Crash dump file**
- Not available

**Video**
- https://youtu.be/WV9_sqYVhWU
- Speaking with the Prusa Help desk they suggested downgrading to the last known good version, so I proceeded to downgrade the printer to version 5.0.1 and this **resolved the issue** :-)
- I first noticed this issue appear in version 5.1.0, released November 23, 2023 and it has steadily gotten worse from that point in time
- I continued to update firmware hoping it would be fixed and started troubleshooting other things such as the USB stick itself in addition to searching forums to see if others were affected, I didn't see many people discussing the issue so I assumed it to be an issue with my machine.
- It seems however downgrading has resolved the issue, so for now if there is anybody else out there experiencing similar 'pausing' behaviour there is a workaround until this issue gets a permanent fix.
- Please note, in my case in order to 'downgrade' I needed to follow special instructions as directed by the Help Desk team: [Click here](https://help.prusa3d.com/article/how-to-update-firmware-mk4-xl_453086)

**Additional system information**
**Prusa Slicer version**
- Version: 2.7.1+win64 (latest version)
- Build: PrusaSlicer-2.7.1+win64-202312121425

**Prusa MK4 firmware version**
- FIRMWARE 5.1.2

**USB stick details**
- ADATA UV250 / 16Gb - FAT32 - Standard USB which came with the machine
- Samsung bar plus 128 - FAT32 - Purchased a brand new drive to verify USB stick wasn't to blame

bkerler, 2024-01-13:
Do you have anything connected to the serial port / xbuddy usb port ?

stuz32, 2024-01-14:
Hey @bkerler ,

I had a USB extension cable plugged in with a freshly formatted FAT32 Samsung bar plus.

The issue however was persistent even with just the USB alone plugged into the machine.

No interaction with the serial port.

Regards,
Stu

Mikyner, 2024-02-05:
I also observe this problem from time to time. All of a sudden the printer stops for about a minute waiting in place and then starts printing again.

stuz32, 2024-02-14:
Hey @Mikyner ,

Thanks for the comment. In order to get around this issue I downgraded my MK4 firmware to version 5.0.1 and I'm no longer affected by the issue.

I'm fairly confident this is firmware related and not the hardware, hopefully downgrading helps you too

Cheers,
Stu

konkrog, 2024-03-08:
Experienced the same issue today on Mk4 with firmware version 5.1.3... The print paused for maybe 30 seconds, and then continued... So far it has happened two times... but I'm not monitoring the printing closely, so it can easily have happened many more times.

K

Prusa-Support, 2024-03-09:
Thanks for reporting.
This may be related to issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3682 which was opened shortly after.
You may want to close this issue in favor of the other one which grew in popularity.
We are investigating.

Michele Moramarco
Prusa Research

lowgos, 2024-03-09:
I do not think it's related to #3682 as the part is not impacted. The MK4 freezes, the screen is not responding to any input either, the fans are still on. And after a while the printing resumes at the same location without issue.
What #3682 describes is something else as the print has a complete part of it omitted.

shledge, 2024-03-19:
Mine just had this exact issue, the first time I experienced it since getting the printer back in May!

stuz32, 2024-03-22:
> Thanks for reporting. This may be related to issue #3682 which was opened shortly after. You may want to close this issue in favor of the other one which grew in popularity. We are investigating.
> 
> Michele Moramarco Prusa Research

Hey @Prusa-Support ,

Yep I agree with @lowgos , I had a look at #3682 but the printing error in this case doesn't look the same.

For this issue, the printer essentially stops printing for a short period of time, and then continues from exactly the same spot and sometimes leaves a small blob / pockmark where it stopped

CommanderRedYT, 2024-04-12:
I experienced this issue just now! For me, the screen still worked. But the printer was stuck on the same place for ~30s. Happened with Mk4 and firmware 5.1.3+13503

Prusa-Support, 2024-05-25:
So far this issue has been as hardly reproducible as issue #3682 and they may have some common causes, potentially related to an empty code buffer, slow data transfer, corrupted bits of the code, external interference ...

I can't tell for sure but I start to believe we may be looking at the same occurrence in different moments of the print or with a slightly different extent.

I can agree on keeping these two issues separated but possibly please follow the indications and findings in the other thread, starting from the suggestions at https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3682#issuecomment-1998040694, including the option to remove any third-party hardware, or even disconnecting the printer from the router.
Any extra piece of feedback or hint for the reproducibility of the problem may come in useful.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-11-30:
The firmware release [6.2.0-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha1) includes overhauled USB media prefetch feature, which should drastically reduce the chance of "clogged buffer" hence "skipped lines/layers".
A later firmware release [6.2.0-alpha2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha2) proved to be a pretty stable firmware over a significant amount of testing hours, providing an effective correction of this erratic bug.
You can rely on the latest alpha release and the final firmware 6.2.0 with a few additional adjustments should follow in short.
 
Michele Moramarco
Prusa Research

closed 2024-11-30 by Prusa-Support