prusa3d/Prusa-Firmware-Buddy #3668: [FEATURE REQUEST] <Custom filament preheat option>

brunofresco, 2024-01-08:
**Printer type** - [Prusa XL]
**Optional upgrades** - [Preheat Menu]

**Is your feature request related to an existing problem? Please describe.**
Because I sometimes use different filaments, I always get errors that the correct filament is not loaded.

**Describe the expected functionality**
There should be a functionality to add custom filaments to the preheat menu and name them, or when it is time to preheat to load a filament, a function called "load custom filament".

bkerler, 2024-01-12:
There's already a PR for that: #3657. However naming them individually won't be possible with the current state of the firmware.

Prusa-Support, 2024-06-28:
This issue can actually be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1258.

Michele Moramarco
Prusa Research

closed 2024-06-28 by Prusa-Support