prusa3d/Prusa-Firmware-Buddy #3669: [BUG] When preheating to unload filament on any tool other than Tool 1, the printer will start heating each tool head.

brunofresco, 2024-01-08:
*Printer type** - [XL]

**Printer firmware version** - [5.1.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [5 Print Heads]

**USB drive or USB/Octoprint**
USB

**Describe the bug**
For example, when preheating for a filament change on tool 3, tool 1 and tool 2 will also heat up.

**How to reproduce**
Just change the filament on any tool and watch the temperatures.

**Expected behavior**
Only heat the tool i need

FiveTeethless, 2024-01-10:
Hi, i can confirm that at least tool 1 at my printer also always heats up to 170, when preheating tool 4 (to PLA).
I saw that several times.
When printing with tool 4 the others are cold, but I can not say, I they were during the heat up and probing phase.

brunofresco, 2024-01-11:
I actually dont know exactly the steps i have done but i noticed that when i preheat for a print i with tool 3 and only tool 3 for print the tool 1 also heats up to 170. sometimes tool 2 to about 80 degrees. Or when I select preheat from the home menu it heats up all the tools.

Today i even noticed after a failed print on tool 3 i left the printer idle for about 6 hours after cancelling the print and the right cooling fan comes on every 30 seconds for about 10 seconds. the temperature of the head was 24 degrees the whole time.

TechnoSwiss, 2024-01-22:
I have this issue as well when changing out filament on any toolhead other than #1, and have also run into it when using the change filament option from the front panel. If I select a tool other than what the model was originally sliced for, the original toolhead seems to heat up (although I've only confirmed this when the original toolhead was #1)

morphias2004, 2024-01-30:
+1   Same issue.

Dorito203, 2024-03-23:
I have the same bug, now over several firmware versions (5 tool). 
I can reliably reproduce the bug, when I change a filament on e.g. E2, E1 is automatically heated up to 170°C after the change. Sometimes this also happens after starting a print where E1 is not used at all.
I often only notice after hours of printing that E1 has been set to 170°C the whole time.

wafflecart, 2024-04-08:
Have same issue. Replication:
- Load filament into tool 1
- Manually set temp of tool 1 to 150c (I use this to stop oozing!)
- Load filament in tool 2
- Tool 1 will heat back up again to 170c

morphias2004, 2024-04-24:
This have been fixed in the production release of FW 6.0.0.

What hasn't been fixed is turning off the heater when the filament change is finished or the toolhead is parked.

They still sit in their docks at full temp till the 10min timout occurs.

closed 2024-07-09 by github-actions[bot]