prusa3d/Prusa-Firmware-Buddy #3670: [BUG] <Enter comprehensive title>

IftiBashir, 2024-01-08:
**Printer type** - MK4

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - Stock Printer

**USB drive or USB/Octoprint**
USB

**Describe the bug**
When printing a model with Grid infill, and plastic is extruded over original infill lines, when the nozzles passes it bumps on the plastic when the grid infill lines have crossed. This is more evident when using IS, causing loud bumps, and if its a small part it can bump the model loose from the bed. 

**How to reproduce**
Print any model with grid infill using IS. 

**Expected behavior**
Never noticed this in the past with standard prints using grid infill and non-IS profile.

mix579, 2024-01-08:
I have seen this even without IS many times, but it's certainly more pronounced with IS. I've seen infill being shredded using PETG and the standard IS speed infill because of those collisions. Why grid is the default is beyond me.

TheRealGreatOldOne, 2024-01-08:
I’m having these issues too. Perimeters and shells - fine. Flawless. When it starts to infill, it all goes to hell - either the extrusion doesn’t stick, nozzle catches and picks up previously deposited material, and I either get collisions causing layer shifts - or huge wads of plastic around the hot end.

In my frustration to solve this I updated to the latest firmware, and the did a re calibration. Which has now caused other issues as my printer will now no longer pass a Y axis check.

All parts / fasteners are tight. The belts pass the Prusa harmonic check for tightness. The pulleys are oriented correctly and fixed in place (no slop between the shaft and the pulley)

This printer is so frustrating. My mk3s was a complete workhorse. Since the upgrade, it’s now more of lame horse.

Zappes, 2024-01-22:
I noticed the same thing with cubic infill, actually. The only infill that seems to not make this kind of a mess is gyroid, but that could simply be because it forces the printer to slow down quite a bit.

closed 2024-07-01 by github-actions[bot]