prusa3d/Prusa-Firmware-Buddy #3671: [ENHANCEMENT] Option to set time manually

anghelos, 2024-01-08:
**Printer type** - MK4

Related: #3523 
There should be an option to set the time manually.
I'm in a school setting where connecting the printers to the WiFi is impossible. Even if I manage to connect them to the internet through Ethernet, the school uses its own time servers and blocks all other requests.

bkerler, 2024-01-12:
You can already set the time manually. Or do you plan to use a local time server ? If so, please change the topic accordingly, thanks :)

mcavoya, 2024-01-12:
@bkerler, please elaborate. @anghelos and I want to manually set the time on our MK4 printers. Is that what you are referring to?

Searching for "mk4 set time" in the Prusa knowlege base yeilds no helpful results.

anghelos, 2024-01-12:
> You can already set the time manually.

How? From what I've found, you can only set the time zone offset, not the actual time.

bkerler, 2024-01-13:
Yes, I was wrong. Sorry for that. I will look into what's possible and try to write a PR.

bkerler, 2024-01-13:
PR #3679 doesn't fully solve your request, but at least you can send the gcode command `M1702 I "192.168.0.1"` to set your own ntp server, which could be set up using an raspberry pi for example. 

I had a look and changing the local time is more challenging as expected in the current firmware, as the time is being taken from the RTC, which is set up using an fixed NTP server on the printer startup. Of course the time could be set manually (in a rather inconvienient scroll bar matter), but the problem would be to somehow override the ntp server and the initialization. I will have a look, hopefully I find a way to sort this out.

anghelos, 2024-01-16:
While your PR looks great, and might somehow solve my problem, it'd be great to also have an option to set the time on printers that will always be offline.

bkerler, 2024-01-22:
The problem is that there is currently no gui method to enter the time / date in a single line approach. It would be possible using multi lines, but wouldn't be very user friendly.

dpark12300, 2024-02-05:
I second this. Maybe using the touchscreen hardware in the mk4?

dpark12300, 2024-02-05:
Or something like the iOS time entry system (roller that can be scrolled with knob)

closed 2024-07-01 by github-actions[bot]