prusa3d/Prusa-Firmware-Buddy #3674: [BUG] <Enter comprehensive title>

FiveTeethless, 2024-01-10:
**Printer type** - [MINI, MK4, XL]
XL

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.0

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
none

**Describe the bug**
I have PLA materials with printingtemperatures lower than 215°C.
The "Move E" item in axis move menu shows "Low temp", eventhough the hotend is still above the 170°C cold extrusion limit.

**How to reproduce**
When loading a filament into the printer only predefined temperatures can be selected.
That is not an issue, as I select the temperature, that fits my material the closest.
Lowering the temperature via settings menu below the preset temperature of 215°C for PLA, makes the extruder not movable from menu anymore.

**Expected behavior**
I want to move the extruder from menu regardless of the "preset" material temperature as long as the temperature is above the cold extrusion limit.

**source-code**
Well I did some digging in the software and found the special material [temperature check](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/6f686bff4e0bb6660815e0bb944b971f1b7577c5/src/gui/screen_menu_move.cpp#L66).
It only alows temperatures within the given and compiled material list of "[filaments](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/6f686bff4e0bb6660815e0bb944b971f1b7577c5/src/common/filament.cpp#L11)".
Like this the absulute minimum extrusion temperature for the machine would be 215°C of PLA.

**my request**
Please open the machine up again for lower temperatures.
I propose different ways of doing so:
- a switch in the settings can disable the material checks (maybe in this [line](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/6f686bff4e0bb6660815e0bb944b971f1b7577c5/src/gui/screen_menu_move.cpp#L159))
- only show warnings instead of "Low temp" (aka DUMMY_AXIS_E) if the temperature is below material temp  
of course the 170°C low temp extrusion check should still be active
- a new material "CUSTOM" will be added to the "[filaments](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/6f686bff4e0bb6660815e0bb944b971f1b7577c5/src/common/filament.cpp#L11)" with the temperature of 170°C  
this solves the move E issue, but would create some challanges during loading or purging with 170°C
- some new feature can actually create a custom filament type with an adjustable temperature  
(of course there must be 5 different within the firmware 😉)

Thank you very much for any review on this issue an keep up your good work.

bkerler, 2024-01-14:
That's no bug, but a feature request. PR #3657 already addresses this issue.

FiveTeethless, 2024-01-14:
Oh, cool, thank you for pointing that solution out.

But I still think the fact that we can not extrude below 215°C with the curent firmware is in fact a bug, as there is the adjustable temperature for low temp extrusion. But it has no effect right now. 😉

If your feature gets accepted at least ther would be a workaround to be abled to extrude also on colder temperatures than 215°C.
I would close this issue as soon as your feature is released.

bkerler, 2024-01-15:
Can you please say which PLA filament you are referring to for temps below 215°C ? Would be good to see the technical datasheet if it is really PLA or a mixture. The 215°C are oriented at the min temp of Prusa Filaments, which is confirmed to work flawlessly on Prusa printers.

FiveTeethless, 2024-01-17:
Hehe 😁 this is an interessting view on the printer eco system.

I've tried it with [Das Filament](https://www.dasfilament.de/filament-spulen/pla-1-75-mm/7/pla-filament-1-75-mm-sonnengelb?c=11) and I have to admit it is quite an old spool.
On many PLA spools you find temps starting at 190°C or 200°C.
(I'm not saying it is the perfect printing temp, but moving the extruder from menu and watching the nozzle when it stops, gives a rough indication of too hot or too cold)

The printer should not at all lock me in to the material from the same manufacturer,
or at least tell me somewhere on the product page, that others are not "supported" if it does.
In my eyes Prusa Research follows strongly the open source thought and wouldn't lock any user into their own eco system. Every indication of doing so is just a bug, that occured during the efforts of making the machine as userfiendly as possible while preserving all the possibilities the machine offers.

If there was a mysterious material that printed at 180°C, I'd like to print and hadle it the same as I would do on my old MK3S.
Of course I'd have to select a tool first 😂

Again, your PR would solve this issue, but until then, it is a bug in the human machine interface. 😉

bkerler, 2024-01-21:
it's not like it's limited to Prusa filament, but it was tested and confirmed to work for these temperatures on most PLA filaments. And for the temps, they have to be hardcoded somehow, so it does make sense Prusa used the temperature of their own filament. Those are my thoughts on that, I do not work for Prusa.

Regarding your filament, it does say "Die Drucktemperatur sollte zwischen 205-230°C liegen", which means that 215°C being in the middle should suit best for this filament as well. Are you sure your thermistor isn't broken / damaged ?

I will have a look if the temp range difference of +-10°C can be added for temperature manual setup.

bkerler, 2024-01-21:
Actually there isn't a real "limitation" in the min temp according to the code [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/0ba6a805f24d08145283d274e1ef79f8455702c7/src/gui/menu_vars.cpp#L88) which allows a starting temperature of 0, as only the max temperature is limited.

I assume that the allowance of +-0.9F [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/6f686bff4e0bb6660815e0bb944b971f1b7577c5/src/gui/screen_menu_move.cpp#L70) for the e-axis is a safety implementation so that users to not try to purge filament if the temperature isn't right, but I will try to ask if it can be increased to allow a higher range.

FiveTeethless, 2024-01-23:
Well you are rigth with the nozzle range not being the limit.
Also as I said, running prints and executing g-codes will actually extrude the filament below the 215°C.

The problem is in the GUI code. The issue is that the current set temperature (+0,9) must be higher than the [material temperature](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/6f686bff4e0bb6660815e0bb944b971f1b7577c5/src/gui/screen_menu_move.cpp#L70C73-L70C103).
If we have the custom or userdefined materials, this restraint is gone.

But funny enough I found another small thing in the GUI code:
it does not actually compare [the temp against the current cold extrusion limit from the marlin core](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/6f686bff4e0bb6660815e0bb944b971f1b7577c5/src/gui/screen_menu_move.cpp#L160), but it compares it against a hard coded [170°C](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/0ba6a805f24d08145283d274e1ef79f8455702c7/src/gui/screen_menu_move.hpp#L80C36-L80C39) 😅
I have already done some adjustments to the marlin server and client, so the GUI can ask marlin for the current cold extrusion, but I'm not sure how to contribute it.
Also I don't want to test it on my machine, as it would void the warranty, but at least it compiles😅

bkerler, 2024-01-23:
You can send a PR over here. Also you can test your PoC using the simulator :)

FiveTeethless, 2024-02-21:
Hi Björn, sorry for the late reply.
Can you please point me in the right direction for the simulator or some "How to".
I might be blind not to find it. 😅
As soon as I tested my code, I'd be happy to share.

FiveTeethless, 2024-07-07:
Well I tried now for 5 hours to get the build environment to work on my Windows 11 machine.
I got the compiler to create the firmware but no simulation to run.
Also in the end the pre-commit hooks were preventing me from committing, as they create some new python venvs in my home direcory, that don't contain the required modules to run successfully.

So I have now a possible small solution to this issue, but still not tested and commited without hooks:
[my Branch](https://github.com/FiveTeethless/Prusa-Firmware-Buddy/tree/use-marlin-cold-extrusion-temp-in-gui)

DoubleStrike, 2024-10-06:
> > some new feature can actually create a custom filament type with an adjustable temperature
> 
> 6.2.0 alpha1 FW has a fully featured user material support https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha1

None of the patch notes refer to fixing the extrusion temp limit, just that you can set custom filament profiles.  Not sure how that would solve this unless that is part of it, but is simply not documented.

CZDanol, 2024-10-06:
From the quick code inspection, the E move should be allowed from 170 degrees: https://github.com/search?q=repo%3Aprusa3d%2FPrusa-Firmware-Buddy%20temp_ok&type=code

FiveTeethless, 2024-10-06:
@bkerler is right, The problem that I have can be solved by simply adding a custom Filament with a Temperature below 215°C.
Like that I can move the extruder from the menu without an issue.

Typing this, I wonder what happens, when I want to extrude below this fixed 170°C. The code snippets that @CZDanol found indicate the "material" must be happy and the temperature must be above the magic constexpr of the screen source code of 170.

To respond to @CZDanol's to statement: No the extruder does not move as the function DUMMY_AXIS_E::IsTargetTempOk() checks the actual temperature against the materials's temperature. This function is good to guide unexperienced users, but restrains experienced users from fine-tuning.

As I said, I tried to get fix it [myselfe](https://github.com/FiveTeethless/Prusa-Firmware-Buddy/tree/use-marlin-cold-extrusion-temp-in-gui), but have no simulation environment to test it.
With this branch I tried to use the actual cold-extrusion-temperature from Marlin Core instead of the magic screen code constexpr. Still the material's temperature would block the movement, but we use the Temperature from the M302 gcode.

I finally was abled to run the simulator on my Windows machine as a MK4. But had no clue, how to send the M302 command to QEMU to actually verify my code worked. But it ran. 😅

FiveTeethless, 2024-12-15:
FW is not released yet

closed 2025-02-24 by github-actions[bot]