prusa3d/Prusa-Firmware-Buddy #3675: [FEATURE REQUEST] Manual mesh bed adjustments

mrhall4, 2024-01-11:
**Printer type** - Prusa MINI+
**Optional upgrades** - Filament Runout Sensor

**Is your feature request related to an existing problem? Please describe.**
The mesh bed values are not fixing the minor variations in the bed and may actually be making it worse. Over and over I have adjusted the X axis to ensure it is perpendicular. The highest point of the bed is in the middle. Both sides of the bed are approximately .003 lower as read by a depth gauge. When printing the mesh bed does not make the proper adjustments to the right side of the print leading to the filament just sitting on top and not being compressed.

**Describe the expected functionality**
When you select the PINDA from the sensor menu you see the values from the previous PINDA probing. You should be able to  adjust the values to compensate for any error in the pinda readings.

WestonMini, 2024-01-11:
I'm having the same problem with my Prusa Mini+. No matter how many times I follow Prusa's squaring and leveling guide the left side is to squished while the right side is barely compressing, I've even attempted to print with the left purposely elevated and still get the same results

agentdouble0, 2024-01-11:
I am also having leveling issues with my Prusa Mini+ where no matter what I do there is a pretty large height difference left to right and this new bed leveling is not fixing it. I followed Prusa's guide on making sure everything is squared and leveled it STILL has that skew. I've tried to print bracket stiffeners as well but nothing seems to fix it. This was NOT an issue before new firmware updates.

The Mini basically always has a difference from left to right due to it only supporting the one side of the X-Axis arm. This is unfortunate but in the past the bed leveling always seemed to fix it well enough to be not an issue but that is no longer the case.

When using Pronterface to view the UBL output the numbers it reads are almost random/constantly changing each time I run it but always with a difference in height from left to right. I won't touch the machine in between UBL's and it still changes a lot but always with a large difference in height left vs right.

It's nearly impossible for me to print anything atm. 

If I could manually change the numbers it's reading and save that bed leveling mesh that would without a doubt fix but as it stands now I cannot.

msd252, 2024-01-30:
Did these issues start on firmware past 4.4.1 for the Mini? I am also having issues with the results of G29 when moving from firmware 4.4.1 to 5.1.0

agentdouble0, 2024-03-29:
> I'm having the same problem with my Prusa Mini+. No matter how many times I follow Prusa's squaring and leveling guide the left side is to squished while the right side is barely compressing, I've even attempted to print with the left purposely elevated and still get the same results

I have the exact same issue. I had no issues with 4.4.1 and even if I revert back to 4.4.1 its fine but with 5.X the bed is NEVER level regardless of what I do.

closed 2024-07-09 by github-actions[bot]