prusa3d/Prusa-Firmware-Buddy #3676: [BUG] Large layer shifts , reproduceable, with a model previously printed fine

beng27, 2024-01-11:
**Printer type** -  MK4

**Printer firmware version** - 5.1.2+13478

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint**
  WiFi over local network 

**Describe the bug**
See attached photo. Normally printing is good on my MK4.  But I'm experiencing large shifts in the XY plane part way through some prints, from a large multi-object project that was generated in a previous version of prusa slicer, but which has been re-saved and resliced in 2.7.1 .  The files  look perfectly good in the slicer and in the g-code viewer, so I am assuming a printer issue rather than a Slicer issue.  If I rotate the prints by 90 degrees on the bed before slicing, they print fine. 

The dislocation is at approximately z=28.35mm   (measured using a vernier caliper)

To be clear, it is not an adhesion problem, the model is well-attached to the build plate.

**How to reproduce**
Send either of the attached files to the printer. 

**Expected behavior**
  See 3 seperate prints in the photo. The left most one is correct & was printed after being rotated 90 degrees in the xy plane. 
  The other two should look the same, ie without a huge displacement of the top part of the model.

**G-code**
  Please attach the G-code you had trouble with.  - bgcodes in "Archive.zip"

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to [reports@prusa3d.com](mailto:reports@prusa3d.com). Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.
[Archive.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13907758/Archive.zip)
[image]

beng27, 2024-01-12:
I should add that the motor pulleys are not loose at all and the belt tensions are correct, so I do not think this is a mechanical issue

bkerler, 2024-01-14:
I had a similar issue and I think the problem might be related to prusaslicer. I used a prusa original model and also any print is fine, except some prints with overhang, which I think is related to this issue. It still might be a mechanical issue on your side as the delayering is massive on the picture. Being curious, am I right that printing the object in parallel to the y-rods the print is fine whereas printing it non-parallel causes the print being messed up ?

TechnoSwiss, 2024-01-22:
I'm having this exact same issue as well, I've seen shifts as large as 20mm in both X and Y. When I contacted Prusa support they think it's a belt tension issue, but I've already verified my belt tension using the tension tower, and 20mm seems like a VERY large shift for a belt tension issue. I've got a part that I've tried printing like 6 times and I've gotten the layer shift 5 times. I know it works, and Prusa support says they can't see anything wrong with the gcode. This is on the XL for me, and I've been trying to get more videos of the failure to try and diagnose because it seems to me like the issues is one of the stepper motors isn't actually making a move when commanded (which would be a single axis shift on the MK4, but both an X-Y shift on the XL because of the Core-XY movement).
[image]

bkerler, 2024-01-22:
@TechnoSwiss the XL is very different to the MK4 regarding those issues. For most people with MK4, tightening the y-belt stronger than the recommended 90hz fixed the issue. For me that seems to have solved the layer issue as well, at least for the last 25 prints.

beng27, 2024-01-22:
I shoudl repeat that my belt tension and pulleys grub screws are all good. There is no possibility that the belt is moving relative to the shaft of that motor. It is really extremely well attached, with no play , slack or looseness.

bkerler, 2024-01-23:
@beng27 Another user said that this happens if the filament sensor doesn't work properly. Can you test print something and then dis/- and reenable the filament sensor to see if you then get a layer shift (should happen more or less immediately) ? Just to rule out issues with the loveboard

mixer3d, 2024-01-29:
Hi i had this issue both on MINI+ and MK4, i wrote more about it in the forum, where you can find more users reporting it too.  One common thing i observed is that printer do that thing after many hours of printing or being switched on, so to mitigate after few series of prints (without any issue at all) i just do power cycle, i thought that it's somehow firmware related, but tricky to reproduce, i managed to make one print which do some warping of PETG, and hotend can hit infill in fast motion. Last layer shift like you descrideb i had few days ago, and hotend moved plate from place... 

Anyway recently i found this regarding MMU3 update:

_Technical updates and resolved issues_

_Layer shifts – In our testing, we had problems with random layer shifts. These were quite rare, but since the print cannot be saved if the problem occurs, they were very annoying to encounter. The prints were clean with nothing the nozzle could crash into._ **The culprit turned out to be a bug in the firmware, occurring only when the MMU unit is connected.** _We believe we have fixed the problem in our latest internal firmware build. We continue to investigate the issue, to make sure the problem is completely eliminated._

I hope this will address also this "random layer shift without MMU".

Source: https://blog.prusa3d.com/development-diary-update-on-mmu3-for-the-mk4_85043/

bkerler, 2024-01-30:
As far as I know this is only related to the mmu and not the printer itself. Anything helping to narrow it down (even videos !) would help.

mressler, 2024-05-10:
I see several posts on the forums related to Layer Shifts. The one that is most detailed can be found here: [Layer Shifts Galore](https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/layer-shifts-galore/) It suggests a possible cause of the layer shift: "Notifications said an endstop was triggered".

[I've had this issue twice with my MK4](https://forum.prusa3d.com/forum/postid/710578/). Once before I had my MMU and once with my MMU connected. In both cases it happened only a few millimeters above the print bed - not as high as the image in this issue report.

Other reports of Layer Shifting:
- [Video of a layer shift in action](https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/annoying-guy-finally-uploads-a-video-showing-his-layer-shifts/) - this one comes with a loud sound as the shift happens. That sounds mechanical to me.
- [Massive Y Axis shift](https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/massive-y-axis-shift/)
- [Layer Shifts](https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/layer-shits/) - I had an issue with this exact model at about the exact same height.

pyrho, 2024-05-14:
Chiming in to say that I'm also randomly experiencing this, feels like it started happening with the switch to BGCODE but that's pure speculation.

mressler, 2024-05-14:
Input Shaper was firmware 5.0.0 and bgcode was firmware 5.1.0 - so they were certainly close in time.

dandrzejewski, 2024-06-18:
I just had this happen on my MINI+ running 6.0.2+14873.  A loud "clunk" and it shifted on the X axis. The only difference between this print and the many others I've done is I had 8 objects on it.  I usually don't print that much at a time.

closed 2024-08-26 by github-actions[bot]

jupe, 2024-10-10:
This started also for my rather new MK4 this week. 3month old printer, none failures earlier, ~3kg printed, mostly PLA, but now random layer shifts without clear reason. Seems to reproduce easier if trying to print many objects.