prusa3d/Prusa-Firmware-Buddy #3678: High power mode and crash detection

johnkray, 2024-01-12:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]

**Is your feature request related to an existing problem? Please describe.**
The MK4 suffers from layer shifts much more than the MK3S+. The MK3 family could recover from minor crashes/layer shifts. This was an essential feature. The MK4 does not have crash detection, thus a small layer shift/crash results in a failed print. Also, the holding torque on the MK4 does not seem to be as strong as on the MK3, likely due to the change to 0.9 deg motors and possibly due to lower motor current settings (not sure about the second one).

I've been operating an on-demand 3D printing service for 7+ years. I've owned 10 MK3s, 5 MK4s, and an XL most all of them facotry assembled. I know what I'm doing and I have extensive experience. The MK4 is great, but it's just a fact that the number of layer shift failures is far greater on the MK4 vs the MK3.

**Describe the expected functionality**
A small crash or layer shift should not result in a failed print. While I would prefer to see crash detection added back into the firmware for the MK4, I have another idea to help alleviate this. I'd like to suggest a high-power mode, which would run the motors at a higher current adding torque and reducing the chance of skipped stepps. This would likely increase the noise level from the motors, but I'm more than happy to make that trade-off for gained reliability, particularly in a production print farm environment. 

Ideally, I'd like to see the high-power mode and crash detection, as I think these features would be very beneficial to print farm operations where reliability is far more important than noise level.

Theheadtwo, 2024-03-20:
This has been a nightmare for me, I can't trust my MK4 to print large things or a bed full of small parts because it will inevitably have a layer shift or multiple shifts.  I'm tempted to throw my MK3 motors back on to make it a MK3.9 or just fully go back to my reliable MK3.  This has been incredibly frustrating.  There needs to be skipped step detection.

m10d, 2024-05-23:
We are also dramatically affected by this.  Good belt tension, decent bed flatness (and MK4 autoleveling working acceptably), zero adhesion issues, enclosure, dried filament. Printing in PETG only which will always have some stringing, especially on parts with lots of bridging or necessary crossing.  

roughly 50% of larger prints are failing partway through with layer shifts in x or y.  Without crash detection, the printer can't even stop, so it wastes a ton of filament before someone goes to check on it.  Which we now must do routinely.  I'm at several kg of wasted filament due to these failures, in just a week of use.  

Our previous primary machine was an MK3, stock w/ default settings - I cannot remember ever having this problem (and can definitively say it was not frequent).

Tupson444, 2024-07-24:
This hasn't happened to me (I haven't printed very large parts on MK4 yet), but I'd like to see this issue addressed. I wonder is crash detection just a delayed in development for MK4, or is it incompatible with the hardware?

Theheadtwo, 2024-07-24:
I'm not sure what's going on with their development, because they also have crash detection disabled on the XL.  However I think the XL has more powerful steppers (or they can push the drivers harder maybe) because I haven't had any issues with crashing on the XL for quite a while.  Meanwhile on the MK4, I'm stuck printing smaller things just to avoid a loss in productivity from having a crash.

I print exclusively with PC blend CF and it can blob up over time on the nozzle or block and can cause crashing randomly.

Tupson444, 2024-09-24:
not stale

rcmaniac25, 2024-11-25:
Still not stale

m10d, 2025-01-27:
Not stale

-------- Original Message --------

rcmaniac25, 2025-05-27:
I've had 3 multi-color print failures in the past month because this doesn't exist. It is very much not stale.

TrianecT-Wouter, 2025-07-28:
Not stale

fekete00, 2025-08-06:
While this doesn’t address the lack of functioning crash detection. I’ve experienced a lot less skipping with ‘Avoid Crossing Perimeters’ disabled, especially on rounded parts. This doesn’t address users not using this feature however

Found in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4637

Tupson444, 2026-09-28:
Crash detection would still be good to have

rcmaniac25, 2026-09-29:
If this existed, It'd have saved me around 300g of PLA and multiple days of restarts and headaches. If it's a hardware limitation, then say so, so I can instead pester support to make a revision of the Buddy board.