prusa3d/Prusa-Firmware-Buddy #3680: MK4 self assembled Y axis 211 error.

Fluffybunny96, 2024-01-13:
This bug is well documented in this forum thread but i cannot see it logged here. 
https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-assembly-and-first-prints-troubleshooting/y-axis-self-test-failing-consistently/paged/2/#post-693944

**Printer type** - [ MK4]

**Printer firmware version** - [5.1.2+13478]

**Original or Custom firmware** - [Original]

**Optional upgrades** - NA

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive/bgcode direct

**Describe the bug**
If y axis bearing is installed as per assembly guide it results in y axis overshooting to 211mm and not giving concise error

**How to reproduce**
Install y axis bearings as per https://help.prusa3d.com/guide/7-y-carriage-heatbed-assembly_438310#458177

**Expected behavior**
  Y axis will overshoot to 211mm (LCD end) and then only give an "x" status on its self text, no other follow up.

**G-code**
na

bkerler, 2024-01-14:
Actually I can't confirm this. I've built my mk4 exactly as stated in the assembly guide and having no issues with version 5.1.2. Can you make a video of the y-axis when the error happens ?

bkerler, 2024-01-14:
Having a look at the firmware, the "211" value comes from this calculation :
[here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/dfc2bc6235505eacee24846e1659d693ed040e80/include/marlin/Configuration_MK4.h#L1124), which is the Y_BED_SIZE being 210mm + 1 mm.

Of course the value of 1 could be increased, but it still seems more to be a mechanical issue and less fw related issue imho.

bkerler, 2024-01-14:
I will do some tests on my printer to see what gap is being replied by my MK4 to get a feeling if the gap really needs to be increased.

Fluffybunny96, 2024-01-14:
So it appears that if your printer is built prior to 5.1.2 this issue doesnt appear and you can downgrade from 5.1.2 to 5.0.0 do all initial self tests and then re upgrade to 5.1.2 the bug doesnt appear(this is described in the forum post.)

[image]
[image]

[Step 19 ](https://help.prusa3d.com/guide/7-y-carriage-heatbed-assembly_438310#458177) shows both bearings must be installed as clost to the centre of the heat bed that the slots allow. This would still work but something in FW has changed the y axis check.

Fluffybunny96, 2024-01-14:
Just commenting to note that this bug may be the same as #3521

Fluffybunny96, 2024-01-15:
@bkerler I also cannot replicate this bug after sucessfully passing the initial y axis test im not sure why but now it has passed once it doesnt want to do it again.

whoyawitt, 2024-01-26:
I have the same issue.  My printer passes calibration with firmware 5.1.0 but fails the y axis calibration in 5.1.2 and 5.1.3.  I have tried adjusting belt tension.  I even used the app and got it into the green.  I also tried shift bearing position as shown in pictures (all the way outside). I cannot get it to work  Will try to move bearings again for the 40th time.

bkerler, 2024-01-26:
I'm now having the same issue with 5.1.3. The solution is simple (for me): Make sure the bed is at the back when turning the printer on and not at the front, then the selftest will pass. But yes, fw issue.

lezetuser, 2024-01-27:
Hi, it happens even to Prusa assembled MK4s see [link](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3521)

lezetuser, 2024-01-27:
@whoyawitt wrote: . For me I had to shift both bearings towards the front. 
When you look at the Y rod, it is longer than the bed Y print length (210 mm). The real Y travel distance is derived from mathematical distraction **rod length** - **the outer distance between the left side bearings**. It does not depend on whether you move both bearings to the front or back.  Closer the bearings are, the travel distance is longer. According to that you have to set the travel distance to stay between the Y travel distance plus the allowed tolerance. 
In the above linked thread is a situation, where the Y calibration for FW versions below 4.7 works, but starting by 5.01 does not work.

whoyawitt, 2024-01-27:
I understand what you are saying and I agree. Mine traveled from -4 to 211 and failed. I moved the bearing to front and used that as a way to stop forward travel and it passed. Instead of moving bearing to front you could do as others have suggested and use zip tie or print a clip to stop forward travel and force movement into the approved range/tolerance. Perhaps something is installed improperly but I checked and rechecked 20 times. I also see in another bug post same issue. Prusa's behavior has been very egotistical. All they have to do is say we understand there is an issue and we will help. For many people with same issue as me it took hours to figure out how to get the machine printing. Take the temporary fixes people have figured out and help others not waste their time instead of insisting there isn't an issue.

DaveTSG, 2024-02-16:
I'm a week into troubleshooting this same problem. I've tried everything mentioned in this discussion. Sadly, nothing has helped. I have nothing to add to this discussion other than "this is frustrating" and "+1".

bkerler, 2024-02-18:
I assume this is a mix of a fw issue (the values being quite tight, although a gap of 2mm should be sufficient) and mechanical issues. It has only happened to me once and it was related to the y-belt in my case. For others it was the position of the bearing, for some they misplaced the plastics on the bearings (or even forgot them). @DaveTSG what values do you get at the y-trave (min and max) ?

DaveTSG, 2024-02-19:
> I assume this is a mix of a fw issue (the values being quite tight, although a gap of 2mm should be sufficient) and mechanical issues. It has only happened to me once and it was related to the y-belt in my case. For others it was the position of the bearing, for some they misplaced the plastics on the bearings (or even forgot them). @DaveTSG what values do you get at the y-travel (min and max) ?

-4mm to +211mm.

bkerler, 2024-02-19:
@DaveTSG Glad to hear you were able to sort it out. Regarding Y-axis layer shifting, check belt and best don't use the app but the Prusament Tension Meter [here](https://www.printables.com/de/model/46639-tension-meter-for-the-gt2-belts-of-i3-mk3s-or-prus) instead.

DaveTSG, 2024-02-21:
> the Prusament Tension Meter

Thanks, but that requires a working printer to allow it to be printed to begin with.

lezetuser, 2024-02-21:
> > the Prusament Tension Meter
> 
> Thanks, but that requires a working printer to allow it to be printed to begin with.

If you tried to tighten the Y bearings due to the failing Y axis test, probably you succeeded to do it too much and there is a high friction between the bearings and the y rod, which could be the cause of your trouble.

DaveTSG, 2024-02-21:
> If you tried to tighten the Y bearings due to the failing Y axis test, probably you succeeded to do it too much and there is a high friction between the bearings and the y rod, which could be the [...]

Actually, I managed to get my printer to work reliably about 12 hours ago. I have since been able to print [the Robo Alpaka model](https://www.printables.com/model/436470-robo-alpaka-sample-model) and [the SpaceX Starship](https://www.printables.com/model/225040-spacex-starship) and they both printed flawlessly.

Actually, is there a good model for stress-testing and ensuring absolutely everything is up and running that I can try to print before declaring victory?

ErwanAliasr1, 2024-02-25:
Solved the same issue with this https://www.printables.com/model/749594-correction-for-y-test-failure-after-firmware-510/comments
That's a bit a pity but worked...

lezetuser, 2024-02-27:
@bkerler 
it is not a separate problem of the Y rod. People are failing also with X axis to home. Sometimes you recognize even 10 ti es bumping to the X rod left side till the printer becomes happy with the result. I would assume something changed also with the stallguard thresholds settings as the number of bumps in X and Y axes as well rapidly increased in comparison with the 4.7x versions.
When you use grafana for the stallguard recording you will find the different values.
btw. you did several bugs corrections, but for many people they are not reachable as they are not able to build the FW from  the source code. When do you expect any new release to let users to get the FW?

rflpazini, 2024-03-11:
I faced the same issue in my MK3.9 with Y axis showing 211mm during the tests. To solve it I downgraded to `5.0.1` ran all self-tests and then came back to `5.1.3`. 
All self-tests were done with the same config and no changes were applied between the versions change!

bkerler, 2024-03-14:
@lezetuser The y-axis is more affected than the x-axis. Most x-axis issues are due to the loveboard cover whereas most y-axis issues are due to the placement of the bearing and the position on printer start. The main problem is that the printer doesn't know its last position before homing (which is rather the way how things work) and the limits are pretty tight in the firmware. Still not sure how to improve that without causing other trouble. You could of course increase the limits but that might cause that build issues aren't detected and print quality won't be as good. 

Regarding the bugfixes and new firmware .. only Prusa knows. Their devs are currently pretty busy, but I'm sure they will have a look. Most gui-related things have to go though a big testing loop, that's why some things take way longer until they are merged.

ErwanAliasr1, 2024-03-14:
Even if there is a tight bearing issue it was working for all of us before
the firmware upgrade and AFAIK no one had print issues because of that.
Couldn't it be acceptable to revert the change instead of us adding a small
part of plastic to make it work ?

bkerler, 2024-03-14:
@ErwanAliasr1 I don't think that reverting the "change" would be helpful at all. Instead I guess the homing routines need to be reworked, although it does seem to work for most users correctly (including me). But I'm sure that Prusa is already working on that, at least they are aware of that issue. I only had the y-axis issue once, that was before doing any homing and was easily solved by just moving the y-bed into the front before self-test. I had a very deep look into the firmware and the spacing of 5mm should be sufficient for most users. If the spacing is higher, it will lead to print issues for sure (due to wrong tensioned belts, wrongly placed bearings, wrongly placed motor holder, defective motors, etc.). In 99% of cases, it's a hardware issue. For 1% of cases, it's indeed a problem with the homing process in the firmware, but that can be solved with moving the bed or homing the printer before doing the self-test. I will try to solve those 1% but so far I wasn't able to reproduce it consistently.

bkerler, 2024-03-15:
@lezetuser I am not working for Prusa, I'm just a passionate coder that tries to help in the spare time. I just occasionally talk to the devs.

lezetuser, 2024-03-16:
@bkerler 
Then my appologies and appreciation. As far as I understand from commits, they are working behind the wall in private folder and from time to time they merge the changes to the public. Not sure they build the .bbf files from the public folder and i guess there are some synchronisation issues as I recently have a build problems with missing files messages. This leads me to the doubts regarding the consistency of the released .bbf files with .bbf files you get by the public folder build.

closed 2024-07-01 by github-actions[bot]