prusa3d/Prusa-Firmware-Buddy #3684: [BUG] MK4 X-Axis Homing Crash

sillily, 2024-01-15:
**Printer type** - MK4 

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - None.

**USB drive or USB/Octoprint**
USB

**Describe the bug**
X-axis attempts to home by first by moving right. If it hits the endstop during this move it tries again. After a number of times doing this the printer crashes. It is the same problem as #3273 but with a MK4.

XY test passes. Nothing is pinching or obstructing the travel to the right. Belts are tuned using the web app. Support has verified the printer's assembly.

This can be avoided by manually moving the extruder left enough for the printer to complete a few full steps right. Unfortunately a lot of actions park the extruder to the far right; finishing a print, load cell self test.

**How to reproduce**

1. Move extruder to far right.
2. Control > Auto Home

**Expected behavior**
Better detection of extreme right x-axis position.

**G-code**
N/A

**Video**
Same behavior in the video in #3273 except it's a MK4.

bkerler, 2024-01-15:
I didn't get a crash on the MK4 at all, but I saw that if you are at the far right on the x-axis and it hits the endstop, the x value increases trying to go right after hitting the endstop every time. It was first -1, then 0, then 1, everytime it is moved right at the endstop and so on. I will have a look. Maybe this is related to this issue.

sillily, 2024-01-16:
Here's an example of the issue after the load cell test is complete. As part of that test it parks the extruder to the far right. When finished it moves on to the z-axis test and starts by doing an x-axis homing. The LCD shows that previously the x-axis test was successful.

The movement isn't as significant as the bug with the mini. However, you can see the belt bounces around quite a bit. Eventually it crashes and reboots.

https://imgur.com/a/b9SG6uq

sillily, 2024-02-04:
Similar issue reported to reddit: https://www.reddit.com/r/prusa3d/comments/1ag0c4a/mk4_xaxis_issue_during_homingsekftest/

Direct link to video: https://www.youtube.com/watch?v=OXmyJeDG2eQ

Note that belt was being pinched by some debris but fixing that didn't fix the crash. They downgraded firmware which then resolved the issue.

bkerler, 2024-02-08:
@sillily The movement on the X-axis that can be seen in the video is NOT normal, could it be that X and Y is switched ? Otherwise you have an issue with the board and you clearly should contact the support.

sillily, 2024-02-08:
@bkerler I presume X and Y are not switched because despite this issue the printer appears to operate normally on these axis. X and Y axis calibration have always passed. What is affecting printability is my z axis is consistently about 0.125mm too low, which I've corrected in the slicer. Is there anything else I should investigate before going to support asking for my board to be replaced?

bkerler, 2024-02-08:
If you disable the motors can you move the nextruder freely with small pressure from left to right ? If not, remove the belt, if it does move freely, make sure that the x-belt isn't tightened too strong. If it still doesn't move freely, check the carriage and also the bearings. If it moves freely, but the belt jumps if you hold the motor pully, then the belt needs to be fastened more.

If this all looks normal, then I assume that your x-axis motor is defective. You can check that if you use the control to move the x-axis. In this case or if you still fail, you should contact support to rule out the cause.

sillily, 2024-02-10:
X-axis freely moves, no belt jumping. Using the Move Axis command there's no issue going full left or right. Belt is "tuned" using the prusa web app with my phone. Prusa support confirmed the setup minus the custom y-carriage.

I factory reset (fw: 5.1.3) and recorded the initial self test: https://www.youtube.com/watch?v=ojDIN2sQ2h0

Sometimes when it gets to load cell test the z-axis binds. Sometimes it doesn't. In this case it happened. It appears the left attempts to keep going while the right is stopped. This does leave skew on the x-axis which I have to correct manually. Once done, I can perform the z-axis test without issue.

bkerler, 2024-02-10:
Looks normal in your video except the z axis. Is there a screw or something in the way ? I need to check the selftest on 5.1.3 and make a video how it normally should look like

sillily, 2024-03-03:
After rebuilding all axis with new parts, I am able to go through the entire test sequence without crashing. There's still a little drama in the z-axis as it raises up for the load cell test. As an aside, the load cell is more consistent on 1st layer. My gut feeling is that there was too much drag on the rods and that threw of some calculations. Keeping the issue open as a crash is never a good solution and it seemed to affect more than me.

closed 2024-07-01 by github-actions[bot]