prusa3d/Prusa-Firmware-Buddy #3688: [QUESTION] Display preview of the print part is disapeart

Wii23Solo, 2024-01-17:
After the firmware update to 5.1.2 on my Prusa mini+, the print part preview is no longer displayed. 

What can I do to get this great feautre back?

With the German firmware 5.1.2 the words on the “more” “repeat” buttons (for filament change) are no longer displayed. 

That was also better before...

GreatLeaderTechnus, 2024-01-19:
English Version of 5.1.2 here, I'm having the same strange occurrence of not having the Print Model Preview...

murk-sy, 2024-02-04:
Which version were you using before?

5.1.0 introduced a different image format for thumbnails, making thumbnails made with older versions of Prusaslicer not appear.

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0

> QOI instead of PNG (XL, MK4, MINI)
>
> Until this release, the thumbnails for G-code were in PNG format, which was rather resource-heavy. We have decided to switch to another format called QOI, which is way less demanding, and we can utilize the freed-up resources to bring more features in future releases.
> 
> **The only downside of removing the support for PNG is that all the G-codes sliced until now, won't have a visible thumbnail on firmware 5.1.0 or newer. For now, the only solution is to reslice the G-code in PrusaSlicer 2.7.0 or newer.**

closed 2024-06-11 by danopernis