prusa3d/Prusa-Firmware-Buddy #3691: [FEATURE REQUEST] <Enter comprehensive title>

KuroiZubon, 2024-01-19:
Hi,
**Printer type** - [MK4]

Feature request: Filament unload procedure automatically trigger cooldown after it done. 

**The problem**
When filament unloading process is done, you have to go to the Control menu then Temperature, then Cooldown and then trigger cooldown. It seems to me that it would be more efficient if cooldown was automatically triggered after filament unloading process.

**Describe the expected functionality**
 That after the filament unloading process is done, the cooldown function is automatically triggered 
or 
add the cooldown option in the filament menu?

Thanks

bkerler, 2024-01-21:
As requested, here is your PR #3700 :)

KuroiZubon, 2024-01-21:
Thank you @bkerler for taking the time to respond to this request!

murk-sy, 2024-02-23:
I believe the default should just be to turn off the nozzle immediately, and that a setting is not required for this.
More info from an issue I opened below.

------------------

### **Is your feature request related to an existing problem? Please describe.**
Recently, the loading/unloading filament functionality has been changed to not heat the bed. However, I believe there is also no reason to continue heating the nozzle after a load/unload.

In my mind, especially considering the quantity of printers in the wild, this may actually also save a non negligible amount of power in the long term.

#### After load:
Heating up the bed will (as far as I remember at least) **always** take longer than heating up the nozzle to ABL temperature. Therefore, keeping the nozzle heated for an extended period is not really worthwhile, in my mind. Some users may want to also clean the nozzle after a load, but I believe that is perfectly doable before the procedure is confirmed completed, or even after while it's cooling down.
Users wanting to preheat the printer while they slice/send gcode would likely want to preheat the bed most of all, which is easily accessible.

#### After unload:
Since "change filament" is an option, the user is unlikely to need the hotend heated after an unload. If a service is planned, the user would either want it cooled to room temperature or at a specific, manually set temperature. The latter would waste a little bit of time at the worst, but the nozzle likely wouldn't even reach the idle temperature in the time the user could manually set it.

### **Describe the expected functionality**
After filament load/unload, instead of 170°C with a 30 minute timeout, the nozzle temperature is set to 0 immediately.

#### If your workflow is different, please share your thoughts.

bkerler, 2024-02-25:
@murk-sy Actually the default workflow is as the firmware has already implemented. The bed is heated up, unloading the filament and immediately loading new filament. It's not in the common workflow to immediately cool down the nozzle as the printer normally would be used. Especially 3d print farms use with scripts and automated systems to unload/load the filament. Further keep in mind that this firmware is used for several printer types that are loaded/unloaded differently (mk3.5/3.9/MK4/XL/Mini). Not having an option switch, this feature will very likely never be added at all to the firmware.

Schwefelholz, 2024-04-25:
Actually, you need to distinguish between the options already provided today:

- Load Filament --> from my perspective it makes sense to keep the nozzle on temperature as it is likely a print is being executed after filament loading
- Change Filament --> again keeping the nozzle on temperature makes sense here, because the user most likely is about to start a print with another filament type/color
- Unload Filament --> this option should initiate a complete cooldown after being executed. If the user chooses to **unload** filament (instead of changing it), it is likely that the printer won't immediately be used for another print.

murk-sy, 2024-04-25:
@bkerler But the bed isn't heated on changing filament, unless that was changed again recently. It used to, which was an annoyance for a while for many users, including me. See https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/706

@Schwefelholz Even with that in mind, I still maintain that not cooling down wastes power and does not save any time.

If the user is changing the filament and planning to start a print:
- If the gcode is already on the printer: The user will immediately start the print. The nozzle has to cool down to ABL temperature anyway, and since the nozzle will not cool down fast enough, setting nozzle to 170°C (or anything around that) is not needed anyway.
- If the user has the print file more or less ready, but still has to send gcode to the printer: The user *should* use preheat menu to start heating the bed *and* the nozzle. Heating the bed takes the longest, and manually starting it when you know you will need it makes sense.
 - If the user does not have the print file ready yet: The user *should not* heat the printer. 

Regardless, my bottom line is that keeping the heater on unnecessarily wastes energy. Additionally, keeping just the nozzle heated without preheating the bed does not benefit starting a print faster - and the bed should not, in my opinion, be preheated when dealing with filament. From the stance of energy saving of course, but also minimising user injury from touching the bed accidentally.
Even assuming the user will start a print afterwards, the vast majority if not all print sheets must be cleaned at room temperature - which I would do after loading filament, cleaning ooze and the nozzle, then the bed, then starting a preheat.

Of course, there are other cases to set/keep the temperature, but those are generally related to maintenance. However, setting temperature/preparing for printing/maintenance should be a part of the preheat menu.
See enhancement https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1795
See bkerler's pull request on improving preheat menu (will probably be implemented) https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3709

murk-sy, 2024-07-02:
Bumping to keep open

Schwefelholz, 2026-09-28:
Still of interest!