prusa3d/Prusa-Firmware-Buddy #3692: [QUESTION] Input shaping enabling/disabling though firmware - Mini+

earglass, 2024-01-19:
Hi!

I would like to know the process on how to go through calibration on Input Shaping in Mini+. I understand there is currently no option in firmware 5.1.2 to turn Input Shaper off, however I'm using own built firmware (due to having Noctua heatbreak fan), so I'm comfortable with changing lines directly in code. I need guidance though, as codebase is pretty messy (although I'm not c++ dev so can't really complain ;)) 
Where can I find lines that would allow me to disable Input Shaping for printer?

Reason - in order to calibrate proper values, I have to print ringing tower without IS. Using older slicer profile doesn't work, ringing tower still prints exactly like with IS profile.

I would like to avoid downgrading to 4.X version, unless there is no other option (seems unlikely).

bkerler, 2024-01-21:
You can use Gcode M593 to set the parameters accordingly in order to disable IS: 
```
M593 X T2 F0 ; disable IS
M593 Y T2 F0 ; disable IS
```
And then use M593 without parameters to see if it is disabled.

See [here](https://forum.prusa3d.com/forum/input-shaping/input-shaper-with-klipper-testprint/) for more details.

earglass, 2024-01-23:
Thank you! I think it would be awesome if Prusa knowledge base had article about this, thanks to re-calibration I was able to print test cube with almost no ghosting at 150mm/s external perimeters :) Gonna tweak it a bit more.

closed 2024-01-23 by earglass