prusa3d/Prusa-Firmware-Buddy #3693: [BUG] <Enter comprehensive title>

Sebbo69, 2024-01-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI, MK4, XL]
XL 5Tools

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.2
**Original or Custom firmware** - [Original/Custom]
original
**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint
PrusaConnect from PrusaSlicer

**Describe the bug**
  A clear and concise description of what the bug is.
When i start a print with Head 5, Head 1 is heating up. The complete print is the Head on full temperature.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

Sebbo69, 2024-01-19:
Now I see the problem in the slicer. but no idea where it comes from. Very little is printed by the first extruder. If extruder 5 is selected then it will not take the first one out of park position at all. But heat it up anyway.

[image]

bkerler, 2024-01-21:
Please write an issue in PrusaSlicer, as it seems to be related with the slicer.

Sebbo69, 2024-01-21:
Okay...

bkerler, 2024-01-21:
Duplicate of #3669. I couldn't see an issue on the firmware implementation, but not 100% sure yet. It's best to go for the PrusaSlicer issue first, as it is most likely gcode generation related.

TechnoSwiss, 2024-01-22:
This isn't just an issue with Prusa slicer, although it IS related to how it's sliced, it might be caused by the same bug as #3669 but could still be it's own issue.

I keep running into this when I slice a model using the "default" toolhead in PrusaSlicer, and then when I print from the front panel I switch out for a different toolhead. It seems like the firmware doesn't quite translate everything over from the default toolhead to whatever was changed to at print-time.

Sebbo69, 2024-01-24:
Problem was my mistake. Printed with wrong profile in slicer. I used the 1 Tool profile and added more extruder.
But now I use the 5Head profile in Prusaslicer and it works

closed 2024-06-11 by danopernis