prusa3d/Prusa-Firmware-Buddy #3694: [FEATURE REQUEST] XL MultiTool menu to show currently loaded filaments and nozzle sizes

tg73, 2024-01-19:
**Printer type** - [XL MultiTool]

**Describe the expected functionality**
Provide a summary view on the screen showing the currently loaded filament and configured nozzle size for each extruder. This would be very useful. It should be easy to access, not buried deep in menus.

bkerler, 2024-01-21:
This is already implemented as footer display option in PR #3601

tg73, 2024-01-21:
I would not find this so useful in the footer, personally. I would like to see a screen like you get when starting a print for mapping tools to different tools - but without having to start a print. The footer items which cycle through all 5 tools are quite tedious and of somewhat limited use in my experience. Maybe they work better for 1 or 2 tool machines.

Anyhow, to use the footer for the purpose I requested would require that 2 footer items are taken up, one to cycle all filament types, and one to cycle all nozzle types. I would not want the footer configured this way during a print. And it would be very tedious to have to configure the footer just to get this info, then configure it back to a sensible arrangement for a print.

I note that #3600 is from a MK3.9 user - so single tool only, where the footer makes more sense.

bkerler, 2024-01-21:
I understand. I will have a thought how this could be implemented, but I doubt it will be easy.

bkerler, 2024-01-21:
How about adding a page in the Info menu called "Extruder Info" ? And showing the current setup of the nozzle diameter + filament type there, similar to the "Sensor Info". That would be quite simple to implement.

tg73, 2024-01-22:
Yes, that would be good. If there's space, it could also show all the other per-extruder data, such as nozzle temp (current and target), heatbreak temp (current and target), fsensor, fan speeds. The temperatures and fan speeds would be particularly useful if this screen can be navigated to during a print.

tg73, 2024-06-11:
Waiting for Prusa to respond.

bkerler, 2024-09-02:
@tg73 sorry, I didn't have yet time to code it, but it's still on my todo