prusa3d/Prusa-Firmware-Buddy #3697: [ENHANCEMENT] Add 1.2mm per-tool nozzle size selection (XL)

tg73, 2024-01-20:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [XL]

**Describe the expected functionality**
Under Settings->Tools->Tool->Nozzle Diameter, only sizes up to 1.0mm can be selected. 1.2mm nozzles are available. Please add 1.2mm to the list.

bkerler, 2024-01-21:
Just wrote PR #3699 for your request.

tg73, 2024-07-02:
Actually, I now own a Slice Engineering GammaMaster 2.4mm nozzle (yes, two point four millimetres). Please support sizes up to 2.4mm.

danopernis, 2024-08-22:
New version of the firmware 6.1.2 should contain the fix, can you test that?

tg73, 2024-08-23:
I can't find where you set the nozzle diameter in 6.1.2. The setting seems to have disappeared. I'll try to get through to live chat and ask them.

tg73, 2024-08-23:
@danopernis testing is pending #4152. But from code inspection, the range appears to be limited to max 1.2mm, so for example the 2.4mm nozzle noted above cannot be properly set up. Please extend the range accordingly.

bkerler, 2024-08-24:
@tg73 I could add 2.4mm as well as a PR, but I don't see that this will make it in the release, as it is no "common" nozzle size and thus won't be officially supported and considered to be a "mod".