prusa3d/Prusa-Firmware-Buddy #3702: [FEATURE REQUEST] Vibration resonance auto calibration

plandevida, 2024-01-21:
**Printer type** - [MINI, MK4, XL]

**Is your feature request related to an existing problem? Please describe.**

Input shaper implementation it is hardcoded or require manual calibration which is complex to achieve and custom GCode, there are also many scenarios where the printer mass can change (sheets, spool holders, mods, extruders…)

Considering the above scenarios and the fact that Prusa is inclined to design machines that can be easily modified input shaping should be an automatic calibration feature performed before every print.

**Describe the expected functionality**

Automatic calibration for the frequency used by input shaper before every print.

This will allow an easy and friendly way to find tune the printer for the user every time something changes on the printer. It will support the community hackers and modders better.

TechnoSwiss, 2024-01-22:
Input shaping should really take advantage of accelerometers in the print head to tune input shaping for each printer installation, and even each head. Take a look at the list of changes that require re-tuning input shaping on a firmware like Kilper, adjusting belt tension, moving the printer to a different work surface, pretty much anything that changes the mass of the print head (ie changed fan ducts, heater blocks, even nozzles...) Input shaping is supposed to improve print quality by compensating for small differences in each individual printer, and one-size-fits-all approach is counter to the whole point. +1

numo68, 2024-03-01:
Auto calibration is not possible without two accelerometers so this is probably unrealistic. Also, calibrating before each print would be not that good idea as it intentionally provokes the resonance, creating forces normally not experienced by the parts. There are however approaches that can be used with or without additional hardware for all of mk3.5/3.9/4 upgrades and the real mk4.

First, Marlin has https://marlinfw.org/tools/input_shaping/freq-calibr.html . It does not work as is (and even messes some settings) and they refused to look at this, stating that it is the Prusa's job to adapt, but the approach is worth pursuing. A simple web-tool could be made available to generate needed gcodes. This is not strictly a firmware thing but I don't know of another github to repo place this suggestion. This would not be the first web-based Prusa tool, see the belt tuner page :)

Second, there are ringing towers available. Again, one might be optimized, declared official and the calibration workflow documented.

Third, while there is no place on the mk3.5 to connect the accelerometer and I am not sure there is the possibility to attach one to the loveboard for the rest (closed source for now but running SPI or I2C over 50 cm along with the motor cable would need at least some shielding), people might have the hardware. The firmware might provide some support in generating movements that could be captured and analyzed offline.

plandevida, 2024-03-01:
Hi @numo68,

Thanks for your contributions.

I agree with you on the calibration process needs the use of accelerometer/s. That could be something Prusa could make more accessible with an upgrade, maybe a love mini board that allows to connect the accelerometer and some hotend cables, so the love mini board can reuse one of the current occupied connectors on the buddy.

On the other hand, I really think it would be a good idea to run that calibration before each print, it will compensate any mass variation, surface change, etc. Take a look to the BambuLab A1 or A1 mini, these printers do it and the results are really good. I don’t think they would have implemented if it wouldn’t give better results.

### For context:
Buddy board connections: https://help.prusa3d.com/guide/5-lcd-assembly-electronics_202703#203404
Buddy board documentation:
https://github.com/prusa3d/Buddy-board-MINI-PCB?tab=readme-ov-file

plandevida, 2024-03-01:
Closed by mistake

plandevida, 2024-03-01:
Closed by mistake

numo68, 2024-03-01:
> ### For context:
> Buddy board documentation: https://github.com/prusa3d/Buddy-board-MINI-PCB?tab=readme-ov-file

Right, the mini Buddy design is available, the xBuddy + loveboard for mk4 not (yet) AFAIK. Looking at the schematics the mini one - if I am not overlooking something - indeed does not export SPI nor I2C to any connector (except the LCD). It also seems to use every such port available (fun fact, the eeprom contains a NFC tag with a connector for an antenna :) ), but I2C is fine with multiple participants and SPI can be solved as well if there are free gpios. Maybe they did export at least the I2C on xBuddy, that remains to be seen.

Swapping cables is an interesting idea and - theoretically - the USB data pins are connected to the right CPU pins and could be used for this. I however kind of doubt Prusa goes for that :)

To me it seems like that they sadly did not plan for this and indirect ways is going to be what we get...

plandevida, 2024-03-01:
Definitely the buddy for the MINI is not designed for such expansions, and probably they would not want to go too hacky here.

if I am not mistaken the WiFi esp32 it is also [ic2](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/#:~:text=The%20ESP32%20has%20two%20I2C,GPIO%2022%20(SCL)), maybe people would sacrifice WiFi over full automatic input shaper.

Racemade, 2024-05-12:
+1 (otherwise hardcoded Input shaping is a nonsense)
While printing, I hold down the Printer for 2 Minutes. You See that Input shaping is hardcoded...
Please do it Like Phase stepping!

Racemade, 2024-07-15:
Stil open..

Racemade, 2024-08-02:
Any news for XL Input shaping calibration? Or are there (official) documentations for the XL to calibrate manually? Ringing tower or something else?

Racemade, 2024-10-05:
Can be closed i think?
Input shaping now exists. Thank you!

Raymogon, 2024-10-26:
Exists now but at least for me it doesnt work, I always get the message that my belts are too lose. Even if I set them way too tight I get this error message.

CZDanol, 2024-10-27:
Well, the feature is indeed implemented. If you have issues @Raymogon , please contact support or create a separate issue with as many details as possible.

closed 2024-10-27 by CZDanol

plandevida, 2024-10-27:
Hi @CZDanol!

Is it not clear to me if the feature implemented is the input shaper with hardcoded values or the auto calibration input shaper, hence it will be included in feature releases.

CZDanol, 2024-10-29:
We have auto calibration input shaper. XL has accelerometers embedded, for MK4, you need to buy an accelerometer accessory: https://www.prusa3d.com/cs/produkt/accelerometer-set/. Mini does not support it because there is no place to connect an accelerometer iirc.