prusa3d/Prusa-Firmware-Buddy #3704: Prusa i3 MK2S with DIY runout sensor source firmware problems....

DeWisz, 2024-01-22:
Hi everybody!
I have i3 MK2S Prusa and decided to add a filament runout sensor. 
I was trying to use directions from here: https://www.thingiverse.com/thing:5136321
I copied all steps directly, but ended up with unresponsive printer with some lit squares on the printer display...
It seams that my computers skills are way below of what is needed to complete this task.
Has anybody a .hex file with firmware that was build in same exact way so I could flash my printer and just wire up my sensor?
Or, is there anybody who could build such firmware for me as stated in link above?

bkerler, 2024-01-24:
Hi, wrong repository I assume ? Here's only for MK4/MK3.9/MINI and XL. If done by mistake, please close this issue and reopen at [here](https://github.com/prusa3d/Prusa-Firmware)

DeWisz, 2024-01-28:
Hi!
Will do. THX!

closed 2024-01-28 by DeWisz