prusa3d/Prusa-Firmware-Buddy #3705: [BUG] using basic authentication causes web interface timeout

twshield, 2024-01-22:
**Printer type** - MINI+

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
   USB flash drive (but not a printing bug)

**Describe the bug**
  Using basic instead of digest authentication causes timeout and no response on web interface.

**How to reproduce**
Here is a command line session (feva is a Fedora 38 box):

Check printer is up (wired interface):
```
feva% ping -c 1 prusamini
PING prusamini.shield.lan (172.30.42.44) 56(84) bytes of data.
64 bytes from prusamini.shield.lan (172.30.42.44): icmp_seq=1 ttl=255 time=0.700 ms

--- prusamini.shield.lan ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.700/0.700/0.700/0.000 ms
```
Use the correct auth method with wrong password, response is immediate:
```
feva% date; curl -i --digest -u 'maker:not_my_pw' http://prusamini/api/v1/info ; date
Mon Jan 22 04:57:20 PM CST 2024
HTTP/1.1 401 Unauthorized
Content-Type: text/plain
Connection: keep-alive
Content-Length: 20
WWW-Authenticate: Digest realm="Printer API", nonce="a9dae69e0000280a", stale=false

HTTP/1.1 401 Unauthorized
Content-Type: text/plain
Connection: keep-alive
Content-Length: 20
WWW-Authenticate: Digest realm="Printer API", nonce="a9dae69e0000280a", stale=false

401: Unauthorized

Mon Jan 22 04:57:20 PM CST 2024
```
Omitting `--digest` so that basic auth is used causes a hang for 13 seconds and nothing returned:
```
feva% date ; curl -i -u 'maker:not_my_pw' http://prusamini/api/v1/info ; date
Mon Jan 22 04:57:36 PM CST 2024
HTTP/1.1 408 Request Timeout

Warning: Binary output can mess up your terminal. Use "--output -" to tell
Warning: curl to output it to your terminal anyway, or consider "--output
Warning: <FILE>" to save to a file.
Mon Jan 22 04:57:49 PM CST 2024
```

Without any authentication response is immediate:
```
feva% date ; curl -i http://prusamini/api/v1/info ; date
Mon Jan 22 04:58:33 PM CST 2024
HTTP/1.1 401 Unauthorized
Content-Type: text/plain
Connection: keep-alive
Content-Length: 20
WWW-Authenticate: Digest realm="Printer API", nonce="a9dae69e00002853", stale=false

401: Unauthorized

Mon Jan 22 04:58:33 PM CST 2024
```

And just for completeness, here is it working with the correct (but XXXX'd out) password (and serial number also XXXX'd):
```
feva% date; curl -i --digest -u 'maker:XXXXXXXXX' http://prusamini/api/v1/info ; date
Mon Jan 22 05:20:11 PM CST 2024
HTTP/1.1 401 Unauthorized
Content-Type: text/plain
Connection: keep-alive
Content-Length: 20
WWW-Authenticate: Digest realm="Printer API", nonce="a9dae69e00002d65", stale=false

HTTP/1.1 200 OK
Content-Type: application/json
Connection: keep-alive
Transfer-Encoding: chunked

{"nozzle_diameter":0.40,"mmu":false,"serial":"XXXXXXXXXX","hostname":"PrusaMINI","min_extrusion_temp":170}Mon Jan 22 05:20:11 PM CST 2024
```

**Expected behavior**
  An immediate 401 response.

**G-code**
  (not a printing bug)

**Crash dump file**
 n/a.

**Video**
  Nothing to see here.

danopernis, 2024-07-04:
@twshield thanks for reporting. Fix is scheduled for next patch version.

danopernis, 2024-07-19:
@twshield New firmware [v6.0.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.4) was just released which should fix this.

closed 2024-08-08 by danopernis