prusa3d/Prusa-Firmware-Buddy #3706: [BUG] USB drive or file error when using Prusa Connect

ErGeeBee, 2024-01-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4
**Printer firmware version** - 5.1.2
**Original or Custom firmware** - Original
**USB drive or USB/Octoprint** - Prusa Connect via Prusaslicer

**Describe the bug**
SOMETIMES when sending a .bgocde to my PrusaConnect MK4 using Prusaslicer, the file downloads on the machine and immediately shows the message 
> USE drive or file error, the print is now paused. Reconnect the drive.
(as seen on first picture)
After clicking continue it displays the screen that print is 100% finished, which isn't the case. (as seen on 2nd picture)

Then, when I choose re-print it prints fine.

**How to reproduce**
Not sure. Cannot find out when it happens and when not. 

**G-code**
[Skoda_Octavia_3_Dashboard_Cable_Clip_0.4n_0.2mm_PLA_MK4IS_10m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14041922/Skoda_Octavia_3_Dashboard_Cable_Clip_0.4n_0.2mm_PLA_MK4IS_10m.bgcode.zip)

**Pictures**
[image]
[image]

bkerler, 2024-01-24:
I think there is a simple fix: Replace the usb drive :D Or make sure that the usb port isn't bent (maybe xlcd issue then).

ErGeeBee, 2024-01-25:
> I think there is a simple fix: Replace the usb drive :D Or make sure that the usb port isn't bent (maybe xlcd issue then).

If I really thought the problem was the usb stick I wouldn't have posted the problem here. Hoping for someone with the same issue. If not I will contact CS to take further steps ;-)

bkerler, 2024-01-26:
Ok, let me check your bgcode file if I can reproduce the issue.

bkerler, 2024-01-26:
I just had a look at it seems the bgcode file itself is already damaged. Is that how the file was stored on the usb-drive or did it come directly from the slicer ?

ErGeeBee, 2024-01-27:
> I just had a look at it seems the bgcode file itself is already damaged. Is that how the file was stored on the usb-drive or did it come directly from the slicer ?

This is the file I downloaded from connect.prusa.com. The file that is stored in my 'Connect Files'

RvO3D, 2024-01-31:
I have the same problem twice with a bgcode generated by the Prusa slicer on MK4, firmware 5.1.2. 
- first uploaded direct to the printer 
- second time, uploaded file from PC to USB.
Seems nothing wrong with the USB stick. I will re-order the stuff and slice it again. See what happens

RvO3D, 2024-01-31:
Even worse, printer crashed after 5 minutes. Will disable bgcode now. Hopefully it helps

[image]

ErGeeBee, 2024-01-31:
> I will re-order the stuff and slice it again. See what happens

What do you mean by re-ordering stuff?

RvO3D, 2024-01-31:
> > I will re-order the stuff and slice it again. See what happens
> 
> What do you mean by re-ordering stuff?

To make sure the slicer starts from scratch.
In addition, the previous failures were at different moments, first crash was after 12% the second at 31%.

jedahan, 2024-02-01:
Do you also have PrusaLink enabled? I can reliably get crashes and bootloops with my USB drive that was working fine before I enabled PrusaLink. Then I disabled PrusaLink and it worked fine. I cannot make a crash dump because plugging in the usb drive is what crashes the system.

ErGeeBee, 2024-02-01:
> Do you also have PrusaLink enabled?
I don't have it enabled

RvO3D, 2024-02-02:
I tried another usb-stick and it works again (no crash). The first usb-stick worked fine (>80 times) before it crashed and kept crashing. With new usb-stick, no problems. I can also not find any issues with the "old"USB-stick with windows and linux checkdisk. I will format again and see what will happen. 
PS other issue: I always have had prusalink enabled.

Sebastian1989101, 2024-02-18:
Don't worry, this issue will come back. I'm now on my 5th USB drive (all from high quality brands except the crappy thing that came with my MK4 - and all working perfectly fine on my PC's and Macs). All had this issue at some point. This is simply an example of how buggy bad the MK4 and XL firmware is as well as PrusaConnect.

RvO3D, 2024-02-18:
I'm using now smaller disks and working now fine with Prusalink. With re-order, i'll make sure slicers start all over.

Sent from Blackberry

bkerler, 2024-02-18:
The problem most probably is with the FAT filesystem. If not properly safely unplugged, there will always be issues. The same applies if data is written on the prusa and the prusa is being resetted.
But of course having a faulty usb stick does also cause issues. It's not easy to track down if this is a fw bug or not, but I'll keep an eye on this issue for sure.

KlausKraemer, 2024-03-21:
The problem occurs a lot and way too many times! (XL 5 Tool)

I don't remove the USB stick. Simply uploading the file is mostly sufficient. Only once in many times the bug occurred two times in succession.

KlausKraemer, 2024-03-23:
> The problem most probably is with the FAT filesystem. If not properly safely unplugged, there will always be issues. The same applies if data is written on the prusa and the prusa is being resetted. [...]

I doubt FAT or a faulty USB stick is the problem. I rarely remove the stick from my XL - have done that exactly once since I got my XL in February. The error occurs around every third file I upload directly from Prusa Slicer via Prusa Connect. I always simply reupload the file without reslicing and thus haven't tested printing with the file that came with the error message.

Maybe I should export the file from Slicer to my harddisk one of next times and compare a MD 5 hash for that file with the one on the stick.

RvO3D, 2024-03-23:
I also never removed the stick (128GB). I'm using a smaller one now (32GB). No issues yet.

Sent from Blackberry

KlausKraemer, 2024-03-26:
Being curious, I changed the Prusa-stick with a 32 GB Sanyo USB 3.0 stick, which never caused any problems. Same result: The bug still shows up. It has to be a FW Problem, a communications error or is it the interface?

MartinMajewski, 2024-05-27:
I'm having the same issue on three different MK4 machines. Sometimes, it works flawlessly, and sometimes, I get this error on every print attempt from Prusa Connect. After dismissing the error message, the XLCDs behave as if the print was finished.

Choosing "reprint" works fine, though.

I am now changing USB thumb drives, as I used SanDisk 64GB USB 3.0 drives on all machines. I will use the stock USB drive from Prusa on one of the machines for testing, now. I will let you know if this issue still happens. If so, is the size or USB 3.0 an issue? I wouldn't be surprised after seeing those lousy network speeds. Maybe Prusa spared some Cents on the USB controller chip, too? 😈

rvalle, 2024-06-13:
I am also experiencing USB error during print with XL 2T, sent via PrusaLink, and FW 6.0.2
It is actually my second USB stick on the XL.

Shushuda, 2024-06-20:
I have this issue as well, but I can reliably tell what causes it in my case.

When I send over bigger files (via Connect) that would take +- over 10 seconds to download if I used LINK instead - no error. Not a single one. I've never had a single error in this case.

When I send over files smaller than that, especially ones that would download near instantly, I have a chance to get this error. I don't know the odds here, but it's working more often than not.

To me it looks like small files that download instantly to the printer are confusing the automatic start while the file would normally still be in the middle of downloading. I guess. It's only ever happening to me when the files are tiny.

LINK works brilliantly and the only difference I can see between it and Connect is starting the print before the download finishes in case of Connect. Maybe this is getting confused when the file gets downloaded too fast?

jbarzy, 2024-06-27:
I just wanted to ping saying that I'm facing the same issue with my Original Mini 6.0.3+14902

I'm sending a sliced file directly from Prusa Slicer straight to my Mini and getting caught usually just over halfway through. My most recent is currently stuck at 91%, big sad.

This is my 3rd USB I've tried as well. Resuming will reheat the nozzle, cause the print head to go into a close position like it's about to start printing, then it just stops and raises the print head to the top most position.

I'm pretty miffed about it as I was originally using super slicer, but after updating my Mini it doesn't want to play nice with super slicer anymore. what a pain.

pgill32, 2024-07-06:
As others have reported, I have experienced the same issue on my Prusa XL.  I had upgraded to the 6.03 version with the resume feature, but found that I hit the USB error much more frequently.  I reverted to 5.1.2+13478 and still get the error, but just occassionally.  Unfortunately, I was printing a nearly 5 day print and hit the error after a day of printing, and a second time after 2+ days of printing.  On the bright side, I was able to power cycle and then resume the print.  On the down-side, the print was using ABS and after the print completed, I there is significant layer separation near the layer being printed at the times of the error.   I am frustrated as the print used 1.7KG of filament and took nearly 5 days to print.  Please make 
 correcting this a priority.

bkerler, 2024-09-04:
> I have this issue as well, but I can reliably tell what causes it in my case.
> 
> When I send over bigger files (via Connect) that would take +- over 15 seconds to download if I used LINK instead - no error. Not a single one. I've never had a single error in this case.
> 
> When I send over files smaller than that, especially ones that would download near instantly, I have a chance to get this error. I don't know the odds here, but it's working more often than not.
> 
> To me it looks like small files that download instantly to the printer are confusing the automatic start while the file would normally still be in the middle of downloading. I guess. It's only ever happening to me when the files are tiny.
> 
> LINK works brilliantly and the only difference I can see between it and Connect is starting the print before the download finishes in case of Connect. Maybe this is getting confused when the file gets downloaded too fast?

Do you use wifi or ethernet ? If you use ethernet, it might be related to an issue with half-duplex vs. duplex, for which I have recently uploaded a possible fix.

bkerler, 2024-09-04:
So we have at least these possible sources causing this issue:
1. Power glitch (issues with power supply or power connector)
2. Shutdown of the printer (in this case, print errors would happen due to filesystem being broken with already existing file)
3. Improper file write using network - connect server vs. half-duplex/duplex (resend issue)
4. Generic issue with writing files leading to filesystem corruption (maybe threading ? multiple file uploads ?)
5. bad usb stick controller

1. and 2. are no fw bugs and cannot be prevented without changing to a journalling filesystem
3. and 4. would be fw bugs which we need to rule out by detailed descriptions
5. also no fw bug, can be fixed by replacing with good usb stick and shouldn't happen fast (except excessive write)

So, let's try to hunt down possible fw bugs by using a proper report:

- Is the printer often powered off between prints ?
- After the issue appeared, did you error check the stick for possible fs corruptions on the usb stick and if yes, did any occur ?
- Does the issue appear often, if yes, when exactly ? (Entering a menu, pressing a button, sending a file, printing with no user interaction or on heating up the bed / nozzle )
- Did you print from usb stick (or menu) or via network ?
- If using network : Did you only transfer one file or multiple ? Did you use PrusaLink or Connect ?
- If using usb stick : Did you plug in the stick before the print happened or did if happen after several prints ?
- If via network, was wifi or ethernet used ?
- Is bgcode or gcode being used as file format and did it happen with large or small filesize (best add filesizes as reference) ?
- Did the screen flicker or does the printer make high pitch noise when starting up or heating ?
- Does a Blue Screen of Death or Red Screen happen ? If yes, please add a screenshot

KlausKraemer, 2024-09-04:
- Is the printer often powered off between prints ?
No
- After the issue appeared, did you error check the stick for possible fs corruptions on the usb stick and if yes, did any occur ?
Once, no error
- Does the issue appear often, if yes, when exactly ? (Entering a menu, pressing a button, sending a file, printing with no user 
interaction or on heating up the bed / nozzle )
Only when I submit a print file via LAN by Prusa Connect
- Did you print from usb stick (or menu) or via network ?
via NW
- If using network : Did you only transfer one file or multiple ? Did you use PrusaLink or Connect ?
Always only one file using Connect
- If using usb stick : Did you plug in the stick before the print happened or did if happen after several prints ?
After prints (randomly), the stick always remains in the printer
- If via network, was wifi or ethernet used ?
Ethernet
- Is bgcode or gcode being used as file format and did it happen with large or small filesize (best add filesizes as reference) ?
bgcode - Tendencially I think smaller files cause more problems, but not sure
- Did the screen flicker or does the printer make high pitch noise when starting up or heating ?
No
- Does a Blue Screen of Death or Red Screen happen ? If yes, please add a screenshot
No

KlausKraemer, 2024-09-05:
Additional info:

1. I changed my USB sticks (Original Prusa) twice. Sandisk USB with 32GB. No better result, error occurs further.
2. My printer is a 5-tool XL
3. The problem occurs rarely 2 times in succession. Mostly the upload after a failed one is successful. Only once I remember three sequential upload errors.
4. The more I think about it, smaller files are more prone to fail.

misc2012-DE, 2024-10-04:
I have the same issue on my MK4s and Prusa XL. I use the delivered USB-Sticks. Sometimes i have this message when i start the print, sometimes the message pops up during printing (just on XL). This is a weird issue and i will not use connect anymore

ChrisHeerschap, 2024-11-11:
Seeing the same on 2 MK4S printers, 1 XL, a Prusa Mini and a MK3.5. USB drives are SanDisk, so known quality, and this is happening across the printers, so I do not believe there to be an issue with the USB sticks. I do believe that the observation by @Shushuda may very well be contributing to mine as well - I just sent a very small print, even with the Generic FLEX filament profile - which is slow - the print is only 4 minutes, and I got the issue. I issued a print last night for a PETG print which was 7 minutes, and saw it there as well. I would strongly suggest pursuing Shushuda's observations about very small files triggering the issue, and I will be watching to see if I get any instances that don't match this pattern.

Mike-FUT, 2024-11-11:
My MK4S has these issues every now and then since the display was replaced and I don't recall ever having that issue with the display that was originally delivered with my MK4S. So I guess that might be hardware (version?) related.
Usually it happens when transferring files via Prusa Connect but also it twice interrupted a print.
Oh, of course I not only tried the Prusa stock delivered USB drive but also a different, brand new Sandisk USB drive, no difference.

CZDanol, 2024-11-13:
If you're having this problem, please try updating to 6.2.0-alpha2. We hope to have mostly fixed this issue in the 6.2 release.

RvO3D, 2025-01-13:
Can be closed. I'm not sure if issue has been resolved or that my workaround works.

Mike-FUT, 2025-01-13:
Since 6.2 is not released yet and not everyone is installing firmware in alpha status this should stay open until 6.2 is released and no issues are reported on 6.2 anymore.
And the USB drive is not the reason for the issues, neither model nor size.

KlausKraemer, 2025-01-13:
I'd really experience the improvement first with release version 6.2 as I don't install alphas or betas.

As soon as the improvement seems to really eradicate this overly annoying bug, it's ok to close the issue.

Prusa-Support, 2025-01-21:
Sure, I understand you may not be keen to try alpha/beta releases but that would be appreciated as a form of collaboration on GitHub, for a more efficient bug-fixing workflow.
Please consider that since 6.2.0-alpha2 was proven to be rather stable for many testing months but now, the _almost final_ version 6.2.0-RC1 is available too.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1
I hope you may be willing to give it a try and provide feedback : )

Michele Moramarco
Prusa Research

ChrisHeerschap, 2025-01-22:
How are we supposed to give feedback on the RC or alpha firmwares? I installed 6.2.0-alpha2 and am now on the RC but I'm not sure where we're supposed to give feedback.

CZDanol, 2025-01-22:
@ChrisHeerschap this is a good place :P

ChrisHeerschap, 2025-01-23:
Not if my feedback isn't pertaining to this particular issue, is it?

CZDanol, 2025-01-23:
Oh, if it's not regarding the USB drive issue, just create a different issue here on github. Just please first check if someone hasn't already reported it before you.

ChrisHeerschap, 2025-01-23:
So there's no general feedback on alpha or RC tickets? Octoprint does this, and I've always used them when running pre-release code even if just to say "everything is working". Doesn't sound like something worth making an issue for.

CZDanol, 2025-01-24:
Well, for general negative feedback, you can simply create an issue and note the correct FW version. For general positive feedback, we currently don't have a channel for that. Might be a good idea, it would be refreshing to read something positive from time to time besides the endless feature requests and bug reports :)

tegbert, 2025-01-25:
My MK4S aborts and nukes bgcode file contents on my usb drive about half the time, i.e., the directory listing reports 0 content for the file after the "abort". I've tried several usb drives with the same results. My firmware is 6.1.4 + 7908. I purchased the MK4S in December 2024 assembled from the factory. I do not use wifi with the machine.

bkerler, 2025-01-25:
@tegbert Does this happen if you store the file directly on the usb-drive or using Link or using Connect ? Also check if the ground cables are properly connected and try to replace the usb stick to see if the usb stick might be faulty.

bkerler, 2025-01-25:
This issue is likely connected to issue #3593 and/or duplicate. People should check this issue for some hints to further hunt down the issue.

tegbert, 2025-03-27:
I have found that if I always unmount the USB drive before removing it, the problem does not occur. It must be some interaction between the USB drive file system and the Linux handler.

murk-sy, 2025-04-25:
Got it again today on 6.2.4-RC1+8908. Didn't do anything particularly weird.
After a print finished, the machine was idle on the main menu. Sent a print (copy to printer only), and connect seemed to have difficulties connecting as it reported "Bug" as error, and the tray icon dimmed and undimmed every few seconds. There was no icon for an active transfer. Turned it off and on again, and it didn't help. Went to manually transfer gcode file and noticed corrupt files again.

ChrisHeerschap, 2025-05-03:
6.2.4-8909 and it's been happening with what seems like increasing frequency lately. Thankfully I can tell it to continue from the PrusaConnect interface, and it does, (so I don't have to go to the printer itself) but it was quite a while without any issues and now has happened three or four times on longer multi-part multi-material prints.

closed 2025-07-09 by github-actions[bot]

jstm88, 2025-08-30:
> However, USB drive/file corruption will continue happening, similarly to other applications, with the exception that `higher reading/writing frequency = higher chances of corruption`. Exposure to heat and magnetic fields may not help either so, ideally, avoid accidentally leaving the USB drive on the magnetic heatbed and keep the LCD outside of enclosures if possible - this may be secondary of importance though.

FWIW, if heat was the problem, we'd see file corruption on USB drives all the time. I see corruption during transfers to my MK4 and it's in a cool environment with the LCD outside the enclosure. And "high read/write frequency" is also not a problem here. The MK4 writes to the drive *extremely* slowly. If anything that should improve its reliability.

> All in all, the outcome of this issue will still be similar to the current one of [#3156](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156), which means that, because USB drive/file corruption can happen, **the CRC check will detect the problem**. As a reminder, the user can recover the print from corrupted files by removing the USB drive (during the print) and replacing the _exact same original file_.

I've also mentioned it on that issue, but catching the corruption only during the print is not a good solution. It should be impossible to write a file to the drive that does not match the data the printer receives from Prusa Connect. The file check needs to be done twice: first (either after the transfer or just before the print starts) to confirm the file was transferred completely, and then incrementally during the print to catch additional errors.

The vast majority of these problems occur due to the printer failing to write the file to the drive. Fixing it at that point by reuploading the file is much easier.

Of course, determining the root cause (i.e. is it bad firmware, or bad hardware) would be the best solution. But ensuring the file is checked in its entirety before printing it for the first time would go a long way towards making the issue less impactful.

murk-sy, 2025-08-30:
I can say that heat is probably not the issue. I have pretty constant 30-35°C in my print room and the few times this has happened it wasn't particularly hot.
Haven't had issues on my Minis, which are in the same room (but use new USB drives). The XL I've had issues with is still using the stock provided USB drive. If I had to guess it's some race condition that not all drives handle gracefully or correctly.