prusa3d/Prusa-Firmware-Buddy #3710: Detect when filament is stuck to spool [MK4][No Upgrades]

hackwin, 2024-01-27:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]
**Optional upgrades** - [No Upgrades]

**Is your feature request related to an existing problem? Please describe.**
Some spools of filament are attached at the end (e.g., Microcenter Inland brand).  The last piece of filament cannot be pulled off spool by the extruder.  The extruder pulls the filament as hard as it can and the tension is very high.  The filament guide gets pulled down and snaps off.

**Describe the expected functionality**
Pause the print if extruder motor is effectively stuck.  And/Or, pause the print if x-axis isn't moving because it's stuck in place.  And/Or, design a filament guide that doesn't snap when pulled downwards.

me25542, 2024-01-27:
that's a good idea. i would like that as well.

blue0ceanby, 2024-01-28:
I could imagine that it is possible via the load cell sensor.
It would be conceivable that the average pressure is determined via the load cell and if there is a strong deviation in the negative direction, there is a blockage across the roller etc. If the limit is exceeded in the positive direction, there will be a jam in the direction of the nozzle.

tkircher, 2024-01-29:
This would also help with tangled filament, where the pull force required becomes extremely high. I had several prints fail on my MK2 due to tangles, where the extruder would just grind on the filament and not 'realize' nothing is coming out, and so it would be trying to print in midair. Seems like the MK4 should be able to detect this kind of failure.

me25542, 2024-02-08:
Maybe have a setting that, when enabled, detects abnormally high extruder current and automatically pauses the print, raises Z all the way up (relieving strain on the filament guide / spool holder), and maybe starts beeping. I can't think of any scenario in which anywhere close to the max extruder current would be drawn, except for filament that is tangled / at the end of the roll, so it _should_ work.

tkircher, 2024-06-10:
Does Prusa have a plan for this?

tkircher, 2024-09-10:
Still want this feature, had a snag in a spool just yesterday that ruined a print.

ventrue3000, 2025-01-22:
I'd like a feature to solve the underlying issue as well. To clarify: The issue is that the filament is stuck on the spool core or elsewhere and does not move. 

The stuck filament detection can theoretically detect this, but in my experience, it mostly gives false positives and still reacts too late when the filament actually is stuck.
And that's the problem: If you try to sense filament movement (or nozzle pressure, which is a function of movement), you'll always be too late. If you want to make sure that the print stays flawless, you have to pause it _before_ the filament gets stuck. The only way I see to do this is with more hardware. 
You could use a second stepper to move filament into a buffer from which the extruder can pull it and pause the print if filling the buffer fails. Or you could watch the spools with a camera and use AI to detect when it's on the last one or two windings..

m10d, 2025-01-27:
this is identical to part of OP, and ~all of the ensuing discussion in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4086

tkircher, 2025-01-27:
Since Prusa will never implement stuck filament detection, you all may want to look at something like this:

https://www.printables.com/model/125850-all-in-one-filament-jam-cleaner-runout-sensor

It's a flexure that bends and triggers an endstop switch when the filament is under tension. With my MK4, I am using octoprint on a RPi 4 and wired the switches to spare GPIO pins. When triggered it pauses the print and runs M300 to buzz an alert.

I've never personally experienced a jam at the nozzle, though the load cell should be able to identify that. I've had dozens of filament jams, enough that I considered building a spool to spool filament rewinder. The thing I linked does fine for both tangles and end of spool events.

m10d, 2025-01-27:
> Since Prusa will never implement stuck filament detection, you all may want to look at something like this:
>
> https://www.printables.com/model/125850-all-in-one-filament-jam-cleaner-runout-sensor [...]

Good link, thanks!  

In industry this is often called a follower or `dancing arm`.  It's can be a very effective solution.  If the device linked is durable & functional, it would be a huge improvement over the "presence only" sensors used on the side of the side of the XL.  I haven't looked up what hardware is used there, I am wondering if the prusa/oem sensor could be "simulated" such that no-filament (as is detected today) and "high tension" could both just report as filament-out.

m10d, 2025-11-05:
> [!NOTE]
> I'm asking prusa to clarify the partially-overlapping issues that keep getting submitted for stuck filament detection problems, so i should move my own comment over here.  The title of this issues pretty clear, to be clear, it's mainly the other issues that need updating.  This continues to cause issues on my `XL5t` w/ latest `fw 6.2.6`

The stuck filament help page does seem (as of 11/2025) to accurately descibe the current implementation:

> It is important to note that detecting stuck filament is possible only inside the Nextruder, as pressure builds up from the nozzle to the upper part of the Nextruder. However, if your filament is tangled on the spool, there will be no internal pressure, just outside tension, which the loadcell won't recognize.

So what prusa has implemented is fundamentally different than this issue.  This issue includes anything _before_ the nextruder assembly, such as excessive friction, or the tail of the filament tied to the spool, cause filament motion to cease in the same way the mentioned tangle would.  The end result seems to be either:

1. the filament fails to extrude (prints to air) for a "while" then breaks outright eventually leading to a filament out and a top surface that can't easily be re-sliced as it's an uneven mess OR
2. For me, it seems to mostly just grind, and the printer happily continues printing into air.

This all all seems strange to me, that Prusa can't detect the tangle-based, as this would seem to be the _easiest_ case. 

Forward force on the nozzle just remains zero, and does not vary with extrusion.  Specifically thanks to inevitable oozing, I believe the force would in fact quickly decay quickly to ~zero.  And clearly it would have never spiked like filament stuck in the nozzle would (that kind of stuck is captured here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4086, as well as many closed [dups](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4323)).    

Due to the nature of partial clogs, irregular filament, filament with high solids content, it would seem much _**more**_ difficult to set a good threshold or heuristic to detect stuck filament based on too HIGH of pressure.  But that's what appears to be implemented.

As such, it must be that prusa found differently than ^ ??  I would really appreciate the dev team enlighten us on what's going on here @Prusa-Support - when the print _fails to fail_ outright it's very difficult to accurate re-slice to the precise layer that failed ... so these are huge amounts of waste.

closed 2026-01-13 by github-actions[bot]