prusa3d/Prusa-Firmware-Buddy #3711: [ENHANCEMENT] Mk4 xBuddy GPIO gcode accessibility (M42, M226 specifically)

Fil4ment, 2024-01-27:
**Printer type** - [MK4]
**Optional upgrades** - [Lack enclosure]

**Is your suggested improvement related to an existing problem? Please describe.**
  I'm trying to build an enclosure temperature control system (arduino) which will be commanded by the G-Code running on the printer, allowing a separate processor controlling chamber temperature, with the printer sending only "desired temperature" in the pre-print code, then waiting for a "temperature reached" response, before continuing to print. I know that I can do this using M42 and M226 commands, which are both supported by the firmware.
  
  But only 3 pins (on a stock Mk4) are defined for "analogWrite" commands, which come from the M42 command: FAN, FAN1, and HEAT0.

**Describe the expected functionality**
  I would like to see some way of setting unused pins to be input, output, input_pullup, or output_pullup (if available in the controller), before addressing them via GCode, for external accessory use.

  There really should be a way of using "M42 P76 T1" to set Pin E12 (unused pin on expansion port) as an output, then run "M42 P76 S128" to drive pin E12 to 50% duty cycle. "M42 P15 T0" to set Pin A15 (TDI(?) on expansion port) as an input, then run "M226 S0" to wait for the arduino to pull the pin low (assuming the arduino is driving the pin high, until temperature is reached).
  
  The M42 and M226 functionality is already in the firmware. You just need to open up the unused pins to user/gcode access.

me25542, 2024-01-27:
I have a very similar plan to Fil4ment, I would also like to see a firmware in which all unused pins (like the ones described here [https://help.prusa3d.com/article/accessory-connectors-mk4_622345](url) are unlocked for the M42 and M226 gcode commands.
Thanks!

bkerler, 2024-01-27:
In fact, it does make more sense to use the specified extension headers in the documentation for mods / third party accessories using the spi or i2c interfaces as recommended in the article and the firmware than the gpios directly for multiple reasons. So to use i2c or spi interfaces as listed [here](https://help.prusa3d.com/article/accessory-connectors-mk4_622345).

Fil4ment, 2024-01-27:
That doesn't list J9, and there's not (as far as I'm aware) an easy gcode command to "wait for a command" over i2c or SPI, and the SPI pins available on J9 aren't usable by me, anyway, as the firmware doesn't allow SPI via Gcode.

It does allow analogWrite via Gcode. It just doesn't have the unused pins defined, or a way to define them.

If I'm missing something, and there's a Gcode command that I can run to send an SPI serial command, and then wait for a reply, I'm all for it, but I'm not seeing it, and GPIO analogWrite would be DEAD simple to implement, if the firmware just allowed unused pins to be defined as PWM or input pins.

rbuehlma, 2024-02-27:
I'm currently running a laser engraver addon on my i3 MK3. This requires a PWM signal to adjust the intensity of the laser head. With the i3, I can use the M42 command to set a free GPIO pin to the requested value. Using I2C would make this much more complex. Would be great if this feature can be implemented.

rbuehlma, 2024-03-18:
> In fact, it does make more sense to use the specified extension headers in the documentation for mods / third party accessories using the spi or i2c interfaces as recommended in the article and the [...]

I looked into the options of using i2c and there seem to be multiple problems there as well:

The GCodes M260 and M261 are not enabled. Trying to enable them by enabling `EXPERIMENTAL_I2CBUS` causes the build to fail due to a naming conflict between the i2c namespace and another declaration of i2c in Marlin.

I2C currently does not seem to be an option.

rbuehlma, 2024-03-19:
I have implemented support for the M260 M261. Details can be found here:

https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3840

michalrudolf, 2024-05-10:
Thanks to @rbuehlma, you can now set up an I2C slave on the unused pin and control it using M260 & M261 gcodes.

closed 2024-05-10 by michalrudolf