prusa3d/Prusa-Firmware-Buddy #3714: [FEATURE REQUEST] Add "Clean nozzle" to Filament menu (XL multi-tool)

tg73, 2024-01-30:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - XL multi-tool, maybe other variants.

**Is your feature request related to an existing problem? Please describe.**
It's very important for the nozzle to be clean before starting a print so that load cell MBL works reliably. In the old days of the MK3, there was often plenty of time and physical access to scrub the nozzle during the normal heat-up routine, but there's no such opportunity with the XL. Also, when the XL decides to start moving, it goes like a beast and will hurt you, so you quickly learn *not* to get involved unless the machine is idle. So, manually cleaning the nozzle(s) before a print can be a tedious procedure, requiring to pick up the right tool, bring the nozzle to the right temp, clean it, park it, set the temp back to zero - for each tool involved in the print. It seems that a "Clean nozzle" action, alongside the other actions like "Unload filament" would be really useful.

**Describe the expected functionality**
A `Clean Nozzle` action in the `Filament` menu would list all tools with loaded filament, and allow the user to select a tool to clean. Upon selection, the tool would be picked, moved to the front of the machine, the bed dropped a bit to give access, and the nozzle bought to temperature. The UI would then wait for the user to indicate they are done, then zero the temp and park the tool. This should be repeatable for other tools without requiring rehoming each time (at least if the machine is not left idle for a while between selections).

mix579, 2024-01-31:
Excellent idea. I've been underwhelmed by the lack of resilience of the load cell and its sensitivity to dirt on the nozzle, with the "nozzle cleaning" at the beginning of the print sheer hit or miss. A way to get the nozzle positioned to properly clean with a brush may help deal with a lot of the "everything-but-perfect-first-layer-unless-it's-PLA-on-a-smooth-sheet" issues I'm seeing

dhyananbho, 2024-01-31:
@mix579 
I use tweezers to clean the nozzles. During nozzle heating, I can use them to clean the nozzle wonderfully and always have a perfect first layer. I already had to relearn this nozzle cleaning with the MK4 and it now works perfectly for me.

mix579, 2024-01-31:
Which I do. For example, with PC-CF I wouldn't be able to make it past the Cleaning Nozzle step without constantly peeling the filament off.

dhyananbho, 2024-01-31:
@mix579 
I use tweezers to clean the nozzles. During nozzle heating, I can use them to clean the nozzle wonderfully and always have a perfect first layer. I already had to relearn this nozzle cleaning with the MK4 and it now works perfectly for me. I mostly print PETG and TPU, no PLA. However, for TPU I had to reduce the preheating temperature in the filament start Gcode from 210°C to 200°C so that no drop of TPU remains at every test point during bed leveling. Perhaps this also needs to be adjusted for PC-CF.

Do you perhaps have a better implementation suggestion for the problem I've described? I think it's more helpful to come up with useful ideas than to be ironic about other ideas.

tg73, 2024-01-31:
For clarity (and given some comments that appear to have been deleted), let me note that the MK4 and Mini, as bedslingers, give easy access to the (one and only) nozzle during the normal heating cycle for the user to clean the nozzle with tweezers. And if the bedslinger starts to move, the extruder will only move at speed in the X axis - so you can position your hand/tweezers fairly safely when cleaning with tweezers. In contrast, the XL is a CoreXY machine, and it can have multiple tools. It tends to keep the active tool near the back of the machine when heating, and quite close to the print bed, which is hard to reach. And if it starts moving, the toolhead can move towards you with great speed and force - it is no joke, and it's not safe to try and reach to the back of the machine and use tweezers. Then there's the issue of multiple tools, and it can be desirable to clean all the tools that will be used in a print. Only the first tool used in a print is picked and heated and used for probing and MBL.

So - this feature request is likely specific to the XL, and may not make much sense to non-XL users.

tg73, 2024-01-31:
I'll also second @mix579 that probing/MBL does not mix will with filaments containing fillers (eg PC-CF and wood-fill). I've suggested allowing a different tool to be used only for probing/MBL, even if it's not used in the print and even if it doesn't have filament loaded - see #3722 and prusa3d/PrusaSlicer#12151.

mix579, 2024-01-31:
Thanks for your explanation.  I didn't really see why my comment was interpreted as ironic. Makes sense now.

nealzip, 2024-03-02:
This does sound like a great feature to add.

tg73, 2024-06-11:
Waiting for Prusa to respond.

fabianmuehlberger, 2025-05-21:
+ 1 for this feature. 
IMHO, this is a must. I the case of using PETG manual nozzle cleaning is mandatory. I am using a .6 HF Hardened nozzle. 
What i usually do after a print finished or before I start a new one. 
1. Pick the tool that I want to clean, and choose purge filament. 
2. wait for filament to be purged + a few minutes to ooze out of the nozzle. 
3. Hit OK to cool down the nozzle 
4. Use a brass wire brush to clean. 

Usually, I do this after I print to prepare for the next job. But an automated feature would be very welcome. Preferably, I would mount a brass or silicone brush in the front center of the build plate (there are already mods for this on printables) and have a fixed routine before and after every print. For printing PETG, cleaning during print would also be beneficial. 
The routine would be some moves (depending on the brush position). 

Best