prusa3d/Prusa-Firmware-Buddy #3715: [BUG] Load Filament to Extruders 2 to 5 will always heat up extruder 1 to 170°C

dhyananbho, 2024-01-30:
**Printer type** - XL

**Printer firmware version**
  XL Firmware Version 5.1.2+13478

**Original or Custom firmware**
  Original

**Optional upgrades**
  Non

**USB drive or USB/Octoprint**
  Ethernet and USB flash drive

**Describe the bug**
  If you select the menu for loading the filament into one of the extruders 2 to 5, extruder 1 is always heated up to 170°C. This makes no sense.

**How to reproduce**
  see above

**Expected behavior**
  Only heat up the extruder that was selected for loading the filament.

**G-code**
  Non

**Crash dump file**
  –

**Video**
  –

DoubleStrike, 2024-02-09:
Just got hit by this again, when loading Tool 5.  Magically, Tool 1 heated to 170 for no reason.

geek5d, 2024-03-05:
To add to this, when loading all 5 tools, starting with 1, load complete, I select load 2, 1 parks and temp sits at 170, good. After 2 loads, and I then move to select 3, I notice 2 stays at the laod temp and does not drop to 170. In order to stop the oozing, I manuall drop 2 to 170, grab the ooze, then load 3, etc. Once the load is successful, can we automatically drop the temp to 170?

ulab, 2024-07-17:
I can not reproduce this in 6.0.3+14902.

I have used the menu to load PETG into Extruder 5 and only this extruder is being heated up.

[image]

(Tool 3 is cooling down after being loaded)

DoubleStrike, 2024-08-21:
This is still happening on 6.0.4+14924.  Any chance this can get bumped up the priority list?  I can see this as a potential fire hazard or injury risk when doing maintenance.

If I have tool 1 undocked and set to the side on a cardboard box for some maintenance, and I go to load filament into another head, an unexpected preheat could be VERY bad.  This is admittedly an edge case, but certainly not an unforseeable one.

closed 2024-10-28 by github-actions[bot]