prusa3d/Prusa-Firmware-Buddy #3720: [BUG] <Print preview wont appear on display>

archetrico, 2024-01-31:
**Printer type** - [MK4]

**Printer firmware version** - [5.1.3]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
  I go through the menu on the printer to print a file, before 5.1 firmware I was able to get a preview of the file before printing, after upgrading to 5.1 I don't anymore

**Expected behavior**
  When I select a file and before pressing "print" I get a preview of the file

tkircher, 2024-01-31:
The image file format was changed. You have to update your slicer.

archetrico, 2024-01-31:
@tkircher Thanks for the comment, but I already have prusaslicer 2.7.1, do you mean i need to re-slice all the files in my usb stick?

tkircher, 2024-01-31:
Yes, any files generated by the old version of the slicer won't have a preview appear.

archetrico, 2024-01-31:
Ok I also use custom printer profiles so I guess that is why even the new files didnt have a preview, I will check if this works after the printer finishes printing and close the toppic. Thanks for the assistance!

patriotik007, 2024-02-01:
I have latest firmware 5.1.3, latest slicer 2.7.1 and do not use custom printer profile.
Even that I have same problem - no print preview available on printer display!
Some hint how to fix it?

archetrico, 2024-02-01:
@patriotik007 good news for you, there is a solution, but double check if you are sure no custom printer profile is being used (make sure it doesnt' say modified) because here is the solution:

-In G-code thumbnails it should write "16x16/QOI, 313x173/QOI, 440x240/QOI, 480x240/QOI, 640x480/PNG"

Since you mention it says something different I think your profile has been modified and its not the original, you can replace that text there and it should work.

Since I replaced that line in my custom printer profile I get the preview like @tkircher said I would.

closed 2024-02-01 by archetrico

patriotik007, 2024-02-05:
> @patriotik007 good news for you, there is a solution, but double check if you are sure no custom printer profile is being used (make sure it doesnt' say modified) because here is the solution:
>
> -In [...]

Hello Archetrico,
thank you for inspiration.

Based on your hint I made this procedure which helps me:
1. Open PrusaSlicer (2.7.1 in my case)
2. Click to tab "Printer Settings"
3. Click to rolete menu with installed printers
4. Chose Add/Remove Printers
5. Then information pop-up window appears "PrusaSlicer is not using newest configuration available ..."
6. Just click "Install"
7. It update printer settings automaticaly
8. Done

So now I see under Vendor version 1.11.11 (before there I had 1.9.1) and in G-code thumbnails are now all formats you described (16x16/QOI, 313x173/QOI, 440x240/QOI, 480x240/QOI, 640x480/PNG)

Thank you for help.

archetrico, 2024-02-05:
That is interesting, my prusaslicer asks me automatically if I want to update those values even if I dont go to that menu or when I upgrade the software... However it is I am glad I could help!