prusa3d/Prusa-Firmware-Buddy #3721: [FEATURE REQUEST] Enable SKEW CORRECTION per Gcode

TRPhoenix61, 2024-01-31:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - Mini, XL, MK3.9/4

**Is your feature request related to an existing problem? Please describe.**
  Is it possible to enable skew correction via G-code for the Mini, XL, and MK3.9/4 models? I am aware that the distortion in the construction of these printers is minimal, but it is still present, as demonstrated in the latest video from "CNC-Kitchen." Many users are likely interested in utilizing this function since it was included in the MK3, but is no longer available in the latest printer models. It would be great if we could continue to use this skew correction function through G-code.

andrem1987, 2024-02-19:
I'd really love to see this feature re-implemented as well. My Prusa Mini has a slightly skewed bed which, somwhow, can't be compensated automatically.

bryn51, 2024-03-01:
I agree this should be addressed. For a printer at this price point compensation/adjustment for geometric issues ought to be featured. Bed levelling is not the end game.

YoannH8437, 2024-05-03:
Necessary and should be open to user.
It is all the point about electronics: it can compensate light mechanic default.
Reaching the equivalent precision without it is more expensive [time and/or money] and less reliable with tightening plastic.
Or propose cnc links between axis as garanty return.

caitlynb, 2024-06-16:
My XL is skewed, and I need skew correction for it.  Large parts (the entire reason to buy an XL) are not coming out square on X/Y.  I agree we absolutely need SW compensation for this.

Prusa-Support, 2024-11-21:
Thanks for reporting.

This problem should find an explanation at the hardware and assembly level so such a feature may overlap and potentially complicate physical printer problems. However, as per issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/385, our product and firmware heads are making considerations about this.

Please allow me to close this issue as a duplicate of the issue mentioned above and kindly, don't underestimate the importance of hardware corrections - ideally, to be carefully evaluated with the help of our Customer Support.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

Tzviatiger, 2025-04-29:
THIS IS SUCH A BIG NEED TBH

Like, i would KILL for this feature, i print very dimensionally accurate parts and even though it's in tolerance, it drives my tism crazy

bryn51, 2025-04-29:
It would be interesting to see what differences may exist between:
a) Mk3 / Mk3S versus Mk4 / Mk4S. < I would expect few if any difference in skew, although the different Frame on a true Mk4 might influence.
b) Mk4/Mk4S versus Core One and XL. < I would expect to see a difference, due to the different geometry.

And yes, surely Skew is a factor stemming from Hardware, not software.  But it could be calibrated and compensated, if only there were a way to do so.  Ideally the printer firmware would calibrate itself for this factor, which would be rather hard for a user to do with no aids available.

closed 2025-08-16 by Prusa-Support