prusa3d/Prusa-Firmware-Buddy #3722: [FEATURE REQUEST] Allow any tool to be used for probing/MBL only (XL multi-tool)

tg73, 2024-01-31:
This is the firmware aspect of prusa3d/PrusaSlicer#12151

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - XL multi-tool

**Is your feature request related to an existing problem? Please describe.**
When doing a print with filament such as wood fill, the relatively chunky additive particles almost always lead to nozzle cleaning failure on the XL, and/or MBL inaccuracy.

**Describe the expected functionality**
In conjunction with [changes to PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/issues/12151), allow any tool to be used only for probing/MBL. There should be no check as to the nozzle diameter, loaded filament type, or whether filament is loaded or not - any tool is valid for probing/MBL use only. If the tool is also used for printing, then the usual nozzle diameter and filament type checks would apply.

tg73, 2024-06-11:
Waiting for Prusa to respond.

robjs1000, 2025-04-17:
Any update to this feature request?