prusa3d/Prusa-Firmware-Buddy #3725: Latest release is source-only

kozross, 2024-02-03:
A new release [has been tagged](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.2.0), but it contains no prebuilt firmware (or any changelogs). Is this intentional or an oversight? As-current, unless you build it yourself, and have a broken appendix, you can't even use this, to say nothing of not knowing which printer(s) have any functionality changed.

murk-sy, 2024-02-04:
This usually happens some time before release, but some other factory may be delaying it (for example, writing up the changelog or updates to other services that must be made before this version can be released).

You can *probably* expect the release within a few days.

kozross, 2024-02-06:
Seems like the release notes and one image have been added. Closing this.

closed 2024-02-06 by kozross