prusa3d/Prusa-Firmware-Buddy #3733: [BUG] 2T Prusa XL constantly crashes on certain prints.

petternilsen, 2024-02-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI, MK4, XL]
2T XL

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.2

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint
USB flash drive, uploaded to via PrusaLink

**Describe the bug**
Certain 2 color prints consistently crashes and reboots the printer.  Typically, it will print fine for a while, then enter into a crash loop where it only prints for a few seconds before crashing and rebooting again.  I only ever got a single blue screen, crash log is attached for that one.  
It appears that the crashes are predominately happening when doing large, sweeping surfaces where the print head is moving fast back and forth.  

**How to reproduce**
No consistant way to reproduce.  But try the bgcode and see the crash log upload.

**Expected behavior**
No crashing.

**G-code**
[stormpooper_0.4n_0.2mm_PLA,PLA_XLIS_1h53m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14215596/stormpooper_0.4n_0.2mm_PLA.PLA_XLIS_1h53m.bgcode.zip)

**Crash dump file**
Crash dump sent to the email address.

**Video**
Too large to upload, view the video here: https://1drv.ms/v/s!ApoK2dV2FmRnh8lM8LykGWLDgziaIw?e=UOKFIT

petternilsen, 2024-02-09:
I tried to upload it as not a binary gcode, no difference. Still crashes hard and constantly when it starts printing on the logo after laying down the background. 
I also tried to restrict it to just a single object in the gcode file, no difference there either.

bkerler, 2024-02-20:
Duplicate of Issue #3664, probably related to usb drive or usb drive implementation

petternilsen, 2024-02-21:
> Duplicate of Issue #3664, probably related to usb drive or usb drive implementation

Except this one crashes hard as opposed to just freezing for a bit, so it does not look like a duplicate.

bkerler, 2024-02-21:
Crash in regular terms would be BSOD, but in this case it just reboots. Could be electronical issue as well. Did you contact support ? @petternilsen

petternilsen, 2024-02-21:
> Crash in regular terms would be BSOD, but in this case it just reboots. Could be electronical issue as well. Did you contact support ? @petternilsen

Most of the time, it just beeped and rebooted the printer, then tried to continue printing, which would fail almost immediately (see the video). I did contact support 11 days ago and attached the only BSOD crash log I got and I also linked to this bug.   I have not received any response yet.  Reference support number is PR27456435842.

petternilsen, 2024-02-21:
> Crash in regular terms would be BSOD, but in this case it just reboots. Could be electronical issue as well. Did you contact support ? @petternilsen

I changed to using a SanDisk USB stick, made no difference, it still crashes and reboots constantly on this print.

Prusa-Support, 2024-03-30:
Thanks for reporting but so far it seems there haven't been similar reported cases.
At this point, I would encourage you to pursue regular troubleshooting with Customer Support, as there is a chance the problem doesn't have to be fixed at the firmware level.

Not sure if you already went through this together with the Customer Support but I'd like to make sure you tried some basic steps including the [**hard reset via menu**](https://help.prusa3d.com/article/factory-reset-mk4-xl_421206).
Yet an alternative could be the **hard reset via bootloader** as per the instructions at https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3629#issuecomment-1998628052.

Something else that I would try is **disabling the crash detection** or **lowering the crash sensitivity**, this way we can check if physical collisions are actually detected and if that plays a role.
Collisions may also happen due to **residues around the nozzle**. If you notice irregular extrusion, clumps on the nozzle, and more oozing than expected, consider drying the filament and ensuring the nozzle is "air-tight" on **the nozzle seal (must not be slanted)** while docked.

Please contact again the Customer Suppor for help going through this and more tests.
The Support agents could escalate the individual case internally and evaluate if the problem may eventually be covered by the warranty.

Michele Moramarco
Prusa Research

petternilsen, 2024-04-01:
Thanks for the response. I did a thorough trouble shooting and the issue seems to have been electrical and a lose connection.

closed 2024-04-01 by petternilsen