prusa3d/Prusa-Firmware-Buddy #3734: [QUESTION] MK3.5 upgrade with Bondtech extruder

bgiot, 2024-02-09:
Can me have a kind of documentation to know what to change in the firmware to support Bondtech extruder if we appply the MK3.5 ugrade on an existing MK3S with the bondtech extruder? or is plan to support it by design ?

Lino77, 2024-02-10:
On the Mk4 MK3.9 and on the mini there is the experimental menu on the MK3.5 just look, I haven't finished my mk3.5 yet but I think it's the same there.
https://help.prusa3d.com/article/experimental-menu-mini-mini_264396

bgiot, 2024-02-10:
covers some of the needed settings.  But the path length will be an issue I think...

Lino77, 2024-02-10:
what you mean with path length

Lino77, 2024-02-10:
Oh well, that never bothered me, the few millimeters, but maybe with the next firmare, because I have a Revo and I won't convert the MK3 before then 

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases#:~:text=E3D%20REVO%20hotend,3rd%20party%20hotends.

jmarianito, 2024-03-08:
Hi,
I have the bondtech extruder LGX and the mosquito, is it possible to install and pass all the tests?
when you include third party.