prusa3d/Prusa-Firmware-Buddy #3738: Park last tool after "Change Filament In All Tools" setting on Prusa XL

bobbyh41, 2024-02-10:
Every time I do a "change filament in all tools" setting, I have to manually go to Control - pick/park tool - park current tool. There is no reason that I see not to automatically park the last tool when you do this option.

MR2C280, 2024-02-29:
I agree, but I would change the request to: "_Add the option to park the current tool after any successful filament load_".  This would save having to exit the Filament menu and then navigate to Control --> Pick Park Tool --> Park Current Tool.  In the case of "change filament in all tools" offer this option after the last change.  

After loading filament I almost always want to park the tool so it doesn't sit there oozing filament as the head cools.

girlontherocks, 2025-09-20:
I want this feature too!